pvvx/ATC_MiThermometer #85: Stopped receiving packets in v2.9

astro-fly, 2021-04-06:
I flashed new sensors with latest v2.9 today, but I could not see the messages in my BLE sniffer.
When then flashed with v2.6 everything worked as normal.

I guess it has to do with:
> 2.9 | Added additional flags to advertising packages
but I did not find info on these flags.

PS. I am using Pi ZeroW with this software to receive packets:
https://github.com/JsBergbau/MiTemperature2

pvvx, 2021-04-07:
This means that this program is not able to parse BLE advertising packages.
https://github.com/pvvx/ATC_MiThermometer/issues/79

fz6, 2021-04-08:
It seems like the changes made in v2.9 also break advertising mode `Atc1441`. I flashed v2.7 instead and now the tool I use (` JsBergbau/MiTemperature2 `) works properly in `--atc` mode.

I think the firmware should be fixed to stick to defined format for `Atc1441` and also use a different UUID for the new custom format you have implemented? Then other utilities can easily differentiate between the different packet formats and parse the appropriate ones.

I appreciate the `custom` format is changing and evolving (and I look forward to trying that out in the future when it's ready and my tools support it), but I think the other existing formats should stick to their definitions? Otherwise it is risky to upgrade.

pvvx, 2021-04-08:
> I think the firmware should be fixed to stick to defined format for `Atc1441` and also use a different UUID for the new custom format you have implemented? Then other utilities can easily differentiate between the different packet formats and parse the appropriate ones.

The format has not changed. An advertisement package has the right to contain several sections. This is standard in BLE.
Sections with UUIDs have not changed. There is no point in fixing the firmware for programs that cannot work with BLE.

pvvx, 2021-04-08:
There is no parsing of the package, but there is some kind of comparison, it is not known with what, as the author wanted:
https://github.com/JsBergbau/MiTemperature2/blob/master/LYWSD03MMC.py#L445-L450

[image]

closed 2021-04-08 by pvvx