pvvx/ATC_MiThermometer #87: New advertising flags: about 8 % shorter battery life?

JsBergbau, 2021-04-08:
Hi @pvvx
thanks for you very energy efficient versions. I've found this very interesting picture https://pvvx.github.io/ATC_MiThermometer/CustPower.html 
As you can see most energy is consumend during advertising.
New advertising with version 2.9 format is 35 bytes long instead of 32 Bytes for ATC1441 Frame. So 3 Bytes more transmitted which means 35 / 32 = 1,09375, so about 9 % longer trasmit time. Now take 1 % off because of baseline current that would mean 8 % shorter battery life. Am I wrong? Is there a possibility to enable old format again?

pvvx, 2021-04-09:
There are more detailed diagrams.
[For a complete event with data updates from the sensor and indicator](https://github.com/pvvx/ATC_MiThermometer/issues/36#issuecomment-770952409)

The typical cycle is roughly the same as the original version. 
[Main loop in Advertising mode](https://pvvx.github.io/ATC_MiThermometer/OriginalPower.html)
But the RF TX level in dB is increased for the custom firmware (+3.01dB).

```
If sensor Normal mode:
If we take a window of 18 ms, TX-RF +3 db, then the average consumption is:
Advertising, no other action ........ 1.85 mA
Advertising + sensor reading and update display ........ 2.85 mA
```
TX sending time is about 300 us.
300 * 3/32 ->  + 28 us * 8 mA
8mA * 0.028ms / 18 ms = 0.0124 mA/ms
Average consumption -> 1.85 + 0.0124
100 * 0.0124 / 1.85 = 0.67 %

pvvx, 2021-04-09:
The main reason for the change is to force script "writers" to different "home system" ones to parse packets according to standard Bluetooth LE rules.

pvvx, 2021-04-09:
Ver 3.0 - Added toggle support for advertising package structures for third-party software written by unqualified authors.

closed 2021-04-09 by pvvx

JsBergbau, 2021-04-09:
Thanks for adding this switch. This also helps to hide the sensors from average users.

BTW: When calling other authors unqualified because they rely on published documentation than you could call another authors also unqualified that advertise data with the same UUID only changing length of the advertisments breaking compatibility with already published data format specification.

pvvx, 2021-04-09:
> BTW: When calling other authors unqualified because they rely on published documentation than you could call another authors also unqualified that advertise data with the same UUID only changing length of the advertisments breaking compatibility with already published data format specification.

I agree - I am not a programmer and I have nothing to claim. Just to make it work right.
The same UUID was created on purpose as most of the scripts I found did not respect the size and gave erroneous values.

pvvx, 2021-04-09:
@JsBergbau - I agree with you and remove the atc1441 format in new version (?)

JsBergbau, 2021-04-09:
Sorry that is a missunderstanding. ATC format is fine. It is most powerefficient because it is shortest. Please keep it.

atc1441, 2021-04-09:
If its just about the size a very short version can be implemented.

Only 3byte MAC
One byte for 8bits of flag, for which values are send
2 byte temp
One byte humi
One byte battery
Plus also only advertise the data on changed values so no counter needed

JsBergbau, 2021-04-09:
> Plus also only advertise the data on changed values so no counter needed

Good idea. On the other hand with devices which have bad reception quality you don't know if data didn't change or it was because of bad reception. Thats why I changed to ATC format only. So every advertisment ATC format is transmitted giving 3times higher chance of receiving packet. 

You do a great job and I really appreciate it. My script was only created because at that time there didn't exist one (or at least I didn't find one easy usable script) and over time it has grown. 

Of course you're right with packet length and different format. I just didn't expect that. There was this UUID I checked and I thought that is enough to prevent accidentally decoding wrong data.

I also like that you've implemented a PIN code proctetion and you and also send MI Like packets and encrypt your data so no one knows for example when you shower. Perhaps someday when I have more freetime I'll include all these features to make them easily usable and uintil that personally I find the ATC1441 format the best tradeoff between information and packet length.

atc1441, 2021-04-09:
Most of these feature did @pvvx  :) 

But thanks still for your words !

pvvx, 2021-04-10:
> Of course you're right with packet length and different format. I just didn't expect that. There was this UUID I checked and I thought that is enough to prevent accidentally decoding wrong data.

If there is no correct parsing of packets, there is no possibility of extension or transition to Bluetooth 5 Extended Advertisements or Periodic Advertisements. Your program will not work correctly.
The format of Advertisements packages has not changed since 2010.

JsBergbau, 2021-04-10:
Thanks for the hint. Changed program to parse the packet only if it is 13 Bytes in size. 

> To reduce the size, the duplicate MAC must be removed from the AD data. MAC are already conveyed in the header.

What do you think about ATC1441_V2 Format? It is the same but without (duplicate) MAC. Instead of starting with `10 16 1A 18` it would start with `0A 16 18 1A` (0A = 10 Bytes Lenth instead of 16) and then directly begins with temperature.

Or we design a even shorter format.
We could also omit battery level in % as we can calculate it from battery level in mV. But don't know if it would be even worth it doing that effort for one byte. Whereas saving 6 Bytes, reduces payload by 46 % and complete packetlength is reduced by about 19 % which is really significant from my point of view.

pvvx, 2021-04-10:
Significantly greater savings can be achieved by using the BT 5.2 standard.
The consumption limit is limited by the leakage in the sleep mode of the control microcircuits LCD, TLSR8251, SHTV3 at the level of 5..8 μA

JsBergbau, 2021-04-11:
> "Expanded advertising" is broadcast on one channel, not 3. This is already a 3-fold gain.

So does the sensor currently transmit (at default interval) every 2,5 on all 3 advertisment channels? Even with that it sometimes still takes some some seconds until all receivers got the packet with a higher sequence number and report it. I thought that is because of eventually listening channel and advertising channel were the same and thus the packet was sucessfully receceived.

> Also, according to BT5.2 there is an automatic adjustment of the RF-TX power level.

That implies that the receiver reports back the received signal strength, right?

So if I got you right it is not worth creating a ATC1441_V2 format with 8 bytes so save energy? 8 Bytes less because I had another idea to save 2 additional bytes.
1 byte for battery percentage level and one byte by reporting the voltage in discrete levels. For example you could define max volatage 3,4V (in my database are about 30 sensors and maximum voltage was 3.34V). 3,4 / 256 = 0,01328125 
So you would have a resolution of  0,01328125 V which should be exact enough. And this 256 discrete steps are also very friendly for influxdb compression.

pvvx, 2021-04-11:
Reception of advertisements on [JDY-10 (TLSR8266)](https://github.com/pvvx/UBIA/tree/master/Sniffer) or CC2540 USB Dongle.
https://github.com/pvvx/UBIA/tree/master/Sniffer/OpenWRT_SnifferAd
Reaches several hundred ad packages per second...
https://www.youtube.com/watch?v=MqFcY5Hovpw

> So does the sensor currently transmit (at default interval) every 2,5 on all 3 advertisment channels? 

Yes.