pvvx/ATC_MiThermometer #88: Question: is there a possibility to display  data from different/external sensor?

msperl, 2021-04-09:
Is there a was to Display data from an external sensor (so data received not transmitted)?

Use case would be an external temperature sensor.

i understand it is about battery life, so maybe a ping pong would be possible:
*send the internal sensor data over BT (just one)
* if there is a response (of some sort) instead in a short time window, than present that data
* possibly alter the presentation showing  the internal and alternative sensor data for x seconds at a time

this obviously requires an external device to transmit the data (I.e a Raspberry Pi or similar)

Just wondering if that is possible/feasible/realistic with the limited battery that the device has.

pvvx, 2021-04-12:
In connection mode, all thermometers with the current firmware consume less.
In connection mode, there are 2 options for display interface.

pvvx, 2022-03-24:
Different models of thermometers have different display buffers. Depends on the type of controller and screen.

0x60 For Xiaomi LYWSD03MMC B1.4 there is an example of clock output: [function lcd_clock()] (https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L1815)

0x22 A unified output for different types of displays and their controllers is presented in the TelinkMiFlasher.html interface itself.
[image]
[function sendExt()](https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L1948)

pvvx, 2022-03-24:
> There would probably be around 100 updates per day, I don't know if that matters much for the battery.

Create a permanent connection and get the temperature and humidity and control the display.
The highest consumption occurs at the beginning of the connection, during the handshake. Another increased consumption occurs if you drop the connection without sending a close command.

seeschloss, 2022-04-15:
Thank you very much for your help, I was able to write an ESPHome display component that displays on my lydwsd03mmc some value it gets from elsewhere using the 0x60 function so I can also write a few letters if necessary (a shame that the top segment for the unit isn't independent! I'm displaying litres so a "L" there would have been nice).

closed 2023-02-18 by pvvx

njbuch, 2024-01-30:
Would it be possible to show only the readings from the external sensor, or does that require a fundamental change in the firmware?