pvvx/ATC_MiThermometer #89: New esphome component for pvvx firmware + "Custom" advertising

airy10, 2021-04-10:
Available here : 
https://github.com/airy10/pvvx_mithermometer

Mainly a copy of the esphome atc_mithermometer component with updated decoding code

pvvx, 2021-04-11:
More relevant is support for Xiaomi Mijia encoding and fix drops when receiving ads in ESP32.
I have not found any examples with [AES CMM decoding](https://github.com/pvvx/ATC_MiThermometer/blob/master/test_adv_key.py) for ESP32.

airy10, 2021-04-11:
Encoding format from the original firmware is already supported by the bundled xiaomi_lywsd03mmc components

pvvx, 2021-04-12:
ESPHome does not work on Android and Windows. Requires specific knowledge, additional costs for unnecessary components and assembly of the system from ordinary users.
In ESPHome It is impossible to solve the simple problem of receiving and parsing advertising packages for the entire BLE group, which has a couple of formats directly on ESP32.

airy10, 2021-04-12:
Well, ESPHome is working great for me as a cheap Wifi bridge for my LYWSD03MMC devices and I just thought this component might be useful LYWSD03MMC for other people too...

pvvx, 2021-04-12:
The component can be useful when fixing ad-packet reception skipping in ESP32 drivers. Otherwise it is not a working solution.

rjblake, 2021-04-12:
> Well, ESPHome is working great for me as a cheap Wifi bridge for my LYWSD03MMC devices and I just thought this component might be useful for other people too...

I can see the benefit when using Home Assistant or other Home Automation. In my case, I need additional ways to listen for advertising packages (home automation server bluetooth does not reach all corners of house) and relay them back to main server; so in my setup (Xiaomi LYWSD03MMC->ESP32 Tasmota->Node Red->Home Automation Server). I have not had any joy with  ESPHome and my ESP32s, so curious if this would work

pvvx, 2021-04-12:
ESPHome, ESP32-wrover-b, 3 sensors are registered.
Half an hour charts. Each graph must contain 180 points (!):
[image]
Log Debug says that only 0.5 AD packets per second were received from all surrounding BLE devices.
The Sniffer says that the loss does not exceed 5% of the packets.

pvvx, 2021-04-12:
> I need additional ways to listen for advertising packages (home automation server bluetooth does not reach all corners of house) and relay them back to main server; 

Reception of advertisements on [JDY-10 (TLSR8266)](https://github.com/pvvx/UBIA/tree/master/Sniffer) or CC2540 USB Dongle.
https://github.com/pvvx/UBIA/tree/master/Sniffer/OpenWRT_SnifferAd
Reaches several hundred ad packages per second...
https://www.youtube.com/watch?v=MqFcY5Hovpw

airy10, 2021-04-12:
Even if packets are lost, even receiving new values for temp and humidity every couple of minutes is good enough for me for that kind of sensors.
I'm using it with Home Assistant for sensors that are too far from my Raspberry Pi.
I can understand that it might not fit some people needs, but it's very much good enough for my needs, especially for the price and easy of use.

pvvx, 2021-04-14:
> Even if packets are lost, even receiving new values for temp and humidity every couple of minutes is good enough for me for that kind of sensors. 

Smart Home users are not interested in sensor readings. They are interested in the execution of scripts, which requires a certain response of timely triggering according to data from the sensors.

rjblake, 2021-04-14:
> Smart Home users are not interested in sensor readings. They are interested in the execution of scripts, which requires a certain response of timely triggering according to data from the sensors. 

Indeed, in the case of my switches and automations. My heating is a different story

> According to the scheme of building a smart home system with ESPHome and BLE, the most unstable component today is ESP32.

Guess I've been lucky, as these have been stable without a reboot/restart for over 4 months now.

> Typical equipment is old cheap smartphones, computers, routers and other household junk.

..or in my case a few thousand Euros of equipment; but understand many HA systems are hacked together bits and pieces that require never ending maintenance to keep running.

pvvx, 2021-04-14:
> Guess I've been lucky, as these have been stable without a reboot/restart for over 4 months now.

This is another advertisement without confirmation.

mr-sneezy, 2021-04-15:
Don't know if anyone cares, but my results with ESPhome on ESP32 have been outstanding with my three reflashed MiThermo sensors. This one is even locked up inside my refrigerator, and I am very interested in the charts recorded by HA to see how my fridge performs....
[image]

rjblake, 2021-04-15:
> > Guess I've been lucky, as these have been stable without a reboot/restart for over 4 months now.
> 
> This is another advertisement without confirmation.
> 
> Xiaomi LYWSD03MMC with custom firmware is able to switch heating without assistance.

[image]
Here is the confirmation. Data at 5 minute frequency written to Domoticz. My heating system is not an On/Off, but a Modulating boiler that is adjusted based on heating requirements for my underfloor heating on the ground floor, wall mounted radiators on other floors and control valves for these, combined with the heat loss/retention properties of the different rooms. So, unfortunately a LYWSD03MMC doesn't meet my needs as the thermostat. I'm certainly not 'advertising ESP32s', as mentioned perhaps I've been lucky so far.

pvvx, 2021-04-15:
Graphs in the thermometer itself are recorded in more detail. And we are talking about the speed of receiving information and without dropout. Actual for PID and more accurate adjustments.

kelchm, 2021-04-17:
I've also observed issues with using ESPHome for receiving BLE advertisements. I've spent a little bit of time trying to fine tune the `scan_parameters`, but the performance is still pretty abysmal.

Fundamentally I think the problem is that an ESP32 can't be listening for BLE advertisements while simultaneously sending/responding to network requests over WiFi. I suppose you could set one up to use ethernet instead, but it seems like a rather roundabout solution.

Here's a graph of the number of advertisements received over a 24 hours period of three of my BLE sensors grouped by periods of 5 minutes. While I certainly don't expect this to be a perfectly flat line, less than one advertisement being captures for a given sensor over a period of five minutes just doesn't work for the use cases I have in mind.

[image]

mr-sneezy, 2021-04-17:
@

> Fundamentally I think the problem is that an ESP32 can't be listening for BLE advertisements while simultaneously sending/responding to network requests over WiFi. I suppose you could set one up to use ethernet instead, but it seems like a rather roundabout solution.

I guess one way to prove where the issue is in the ESP32 system is combine an ESP32 with a ESP8266 (or second ESP32), and send the received data from the ESP32 listener (without any WiFi functions) to the ESP8266 via serial for the ESP8266 to send to HA (or wherever the target is). Could use MQTT instead of ESPhome service as well...

pvvx, 2021-04-17:
> I guess one way to prove where the issue is in the ESP32 system is combine an ESP32 with a ESP8266 (or second ESP32), and send the received data from the ESP32 listener (without any WiFi functions) [...]

Questions about missing advertise packages for ESP32 are available on many Internet resources.
The selection of scan intervals does not help.

https://github.com/pvvx/ATC_MiThermometer/issues/59

pvvx, 2021-04-17:
> Or is there a better HW just as cheap as an ESP32 we could use ?

Xiaomi LYWSD03MMC. TLSR8251 has USB2.0 FS. USB pins are available on the board...
JDY-10 (TLSR8266) = $1
BW16 (RTL8720DN) BLE + WiFi 5GHz, USB2.0 HS 
...

airy10, 2021-04-17:
Note that while the ESP32 will never be the device to use if you need high availability for BT+Wifi, you can improve BLE adv reception by enlarging the scanning window, which is very small by default. I'm getting about 35% of them with these parameters in a quite noisy environment - which is enough for what I'm using it for - you'll decide if it is good enough for you :) 
`esp32_ble_tracker:
  scan_parameters:
    active: false
    interval: 211ms
    window: 120ms
`
Default is 350ms/30ms
(using it my LYWSD03MMC using 10s advertising interval/ and x10 measure interval - which is much less that the default setting but I don't need more points)
I wouldn't use that for life-saving features, but it's definitively good enough for me for remote checking temperatures which aren't changing quickly).

pvvx, 2021-04-17:
What prevents the RF receiver in ESP32 from working permanently? The CPU does not take part in the reception. Usually, a packet received by the radio path causes an interrupt and is processed by DMA.

kelchm, 2021-04-18:
I appreciate the insights @pvvx -- honestly this isn't something I've had much time to research thoroughly.

Are there any "plug and play" solutions that could be either used with MQTT or easily adapted to use with Home Assistant? Not that I don't like tinkering with these things, but my time is at a premium right now. Part of the appeal of ESPHome was the fact that the time it took to set it up was trivial.

elRadix, 2021-04-18:
@kelchm you can use the customer component below to integrate your devices with home assistant. It works great.

https://github.com/custom-components/ble_monitor

zdebel, 2021-04-19:
[image]
This is from my custom setup, I use my own sketch on ESP32 for BLE -> MQTT deal, influxdb for database and grafana for presentation. I also log via uart each packet, I get one every few seconds. What bothers me more, is the fact that your firmware drains the battery faster (default settings) than the 'original' atc1441 as can be seen here:
[image]
I confirmed it using my multimeter.

zdebel, 2021-04-19:
> What prevents the RF receiver in ESP32 from working permanently? The CPU does not take part in the reception. Usually, a packet received by the radio path causes an interrupt and is processed by DMA.

From what I understand, WiFi and BLE are multiplexed, thus you can't be in 'BLE rx' mode all the time.

stevoh6, 2021-04-19:
One important note, why is this important:  ESPhome + temp sensor (as those ble ) is perfect solution for smart automation, especialy heating. ESPhome has native support of checking availabilty of HomeAssistant ( whole automations can be run in super "smart" way, HomeAssistant+NodeRed+whatever you want) but in case of power outage, server failure etc, ESPhome can automaticly "switch" to safe mode and act as regular thermostat (maybe little bit smarter with some simple automations written directly in esp). Im using this "safe" scenario even for wall switches.

pvvx, 2021-04-20:
> [image]
> This is from my custom setup, I use my own sketch on ESP32 for BLE -> MQTT deal, influxdb for database and grafana for presentation. I also log via uart each packet, I get one every few seconds. 

Your graph should have 1080 points in 3 hours. At default settings, the data changes every 10 seconds, but this is not on your graph. 80% of packages are missing.
3 hours x 60 minutes 60 seconds in 10 second increments creates 1080 points. And on the presented chart there are only about 160 points in 3 hours. :)

>What bothers me more, is the fact that your firmware drains the battery faster (default settings) than the 'original' atc1441 as can be seen here:
> [image]
> I confirmed it using my multimeter.

Is the battery charging in the center of the graph? :)

pvvx, 2021-04-20:
> From what I understand, WiFi and BLE are multiplexed, thus you can't be in 'BLE rx' mode all the time.

Testing was carried out with the pin in the UART. WiFi is off.

pvvx, 2021-04-20:
MHO-C401 records from Flash Memo. Step 10 minutes.
[image]
[image]
The battery voltage always repeats the temperature schedule.

zdebel, 2021-04-20:
> Your graph should have 1080 points in 3 hours. At default settings, the data changes every 10 seconds, but this is not on your graph. 80% of packages are missing.
> 3 hours x 60 minutes 60 seconds in [...]

[image]
I did a query on my db, around 400 points every hour. Disabling logging to flash made all the difference in regard to power consumption. Thanks for your hard work man, all this makes the device very attractive.

pvvx, 2021-04-20:
> I did a query on my db, around 400 points every hour. 

If your script does not take into account the change of the metering counter, then for an hour there should be 60 * 60 / 2.5 = 1440 received AD packets.
If we take into account that there are other RF sources operating at 2.4 GHz between the sensor and the receiver, then the number of beaten packets will be at the level of several percent...

> Disabling logging to flash made all the difference in regard to power consumption. 

Disabling the recording of measurements in Flash will give no more than 1% battery savings.

camaz, 2021-04-20:
As someone who also uses HA (Homeassistant) withy ESPHome, I can say I also experienced degraded ADV RX performance on an ESP32 (M5Stack Atom Lite); somewhere above 80% packet loss.

I have switched to a raspberry pi zero w with BlueZ and (the unfortunately deprecated) HCITool. I now get just 10~20% packet loss, depending on connection distance.

Many thanks for the project PVVX!

deepcoder, 2021-04-21:
@camaz I too struggled with bluez and python ble on rpi and ubuntu. I wrote a rather rough c app that runs on both rpi and intel ubuntu. I collect from 20 sensors with good success for going on 6 months now. I don't collect missed packet data as I collect from govee sensors as well as xiaomi. That said from sensors that update sub-minute I see those sensor reading consistently. Perhaps it will give you some ideas : https://github.com/deepcoder/bluetooth-temperature-sensors

pvvx, 2021-05-23:
[image]
[image]
[image]
https://github.com/pvvx/ble_monitor
NanoPi R1, NanoPi Neo Core2 + USB BT, ...

metronidazole, 2021-06-24:
Thanks - was wondering why I couldn't get any sensor data using this firmware. Any plans to get (attempt) this merged with upstream?

maikvitesse, 2021-06-26:
> Thanks - was wondering why I couldn't get any sensor data using this firmware. Any plans to get (attempt) this merged with upstream?

I'm having the same problem here.
Flashed this custom firmware from pvvx (thanks!!), created a config for my ESP32 with ESP home, but all my sensors won't report to Home Assistant.
The value keeps saying 'unkown' as actual value.

My ESP-Home config:
`esphome:
  name: ble-receiver
  platform: ESP32
  board: lolin_d32_pro

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "SomePassword"

wifi:
  ssid: "SomeSSID"
  password: "SomePassword"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ble-Receiver Fallback Hotspot"
    password: "SomePassword"

captive_portal:

esp32_ble_tracker:

sensor:
  - platform: atc_mithermometer
    mac_address: "CorrectMAC"
    temperature:
      name: "Slaapkamer - Temperatuur"
    humidity:
      name: "Slaapkamer - Vochtigheid"
    battery_level:
      name: "Slaapkamer - Batterijniveau"
  - platform: atc_mithermometer
    mac_address: "CorrectMAC"
    temperature:
      name: "Woonkamer - Temperatuur"
    humidity:
      name: "Woonkamer - Vochtigheid"
    battery_level:
      name: "Woonkamer - Batterijniveau"
  - platform: atc_mithermometer
    mac_address: "CorrectMAC"
    temperature:
      name: "Kleedkamer - Temperatuur"
    humidity:
      name: "Kleedkamer - Vochtigheid"
    battery_level:
      name: "Kleedkamer - Batterijniveau"
  - platform: atc_mithermometer
    mac_address: "CorrectMAC"
    temperature:
      name: "Badkamer - Temperatuur"
    humidity:
      name: "Badkamer - Vochtigheid"
    battery_level:
      name: "Badkamer - Batterijniveau"`

Any hints? Bluetooth advertising is set to 'custom'

jddonovan, 2021-11-09:
Just as a side note, the C3-versions of the ESP32 chips seem to have a much better BT stack and hardware. There's still around 15% packet loss, but this already means that with 4x broadcast per value the drop rate is inconsequential. It might be possible to drop the package loss further by optimising the code on top of the stack.

kelchm, 2021-11-10:
@jddonovan that's great news, thanks for sharing. I will definitely have to order some to give it a try.

I spent some time trying to get a scanner up and running on the Wio Terminal (which has the ATSAMD51P19 + dedicated RTL8720DN for bluetooth + wifi), but never got it fully working. I think there are some bugs in the libraries provided by Seed studio.

jddonovan, 2021-11-10:
@kelchm just be warned that support for C3 is still limited. Esphome is practically unusable: using I2C oled display freezes the boar after around 15 mins (memory leak?), bluetooth stack and MQTT support is missing and PWM output silently fails.

My tests were with using Arduino, the readily available BT stack and MQTT. I had no issues with this simple setup. Your mileage may vary though. :)

pvvx, 2021-11-11:
USB-BT and other BT adapters on Windows do not accept BLE ads. But you can do it on a cheap JDY-10 module or similar - [AdScanerTrg](https://github.com/pvvx/AdScanerTrg) - BLE ADV repeater. When connected to AdScanerTrg, it transmits ad packets from other devices. The losses are minimal, less than that of the ESPxxxx.

pvvx, 2021-11-11:
> I spent some time trying to get a scanner up and running on the Wio Terminal (which has the ATSAMD51P19 + dedicated RTL8720DN for bluetooth + wifi), but never got it fully working. I think there are some bugs in the libraries provided by Seed studio.

Seed studio uses the interface on SPI and after fixing to UART (reverse recovery in AT-files from files from SDK Ameba) everything works successfully.

closed 2023-02-18 by pvvx