pvvx/ATC_MiThermometer #91: Kalman filter for temperature readings

gazzat5, 2021-04-14:
Hi there, thanks for the great work on this firmware!

Just wondering if you might be able to implement a 'kalman filter' in the firmware to improve temperature readings by using the state history to improve future readings and more closely estimating the actual temperature. This would reduce the flipping between temperature readings when it's close to the next value but not quite there. 

See attached graph for what I'm talking about with regards to the values jumping between two points. 
[image]
Cheers
Garreth

pvvx, 2021-04-15:
Sensor noise in the range of about + 25C:
[image]
The transmission is carried out with a resolution of 0.01C.

pvvx, 2021-05-26:
There is an ad format with an accuracy of 0.01 units and no filters are required.
[image]

closed 2021-05-26 by pvvx