pvvx/ATC_MiThermometer #92: Parsing data with Node-Red

DImagine, 2021-04-14:
Hi! I'm trying to get information via Node-Red using node-red-contrib-generic-ble node set. I can receive information with "Generic BLE in" nod from LYWSD03MMC with custom firmware. I'm able to find humidity and temperature info (it is in bold below), but I cannot find battery level. Can anyone help me with this?

Data received looks like this:
{"uuid":"a4c1388c48b6","characteristics":{"2a00":[65,84,67,95,56,67,52,56,66,54],"2a01":[0,0],"2a04":[16,0,16,0,124,0,232,3],"2a24":[76,89,87,83,68,48,51,77,77,67,0],"2a25":[70,49,46,48,45,67,70,77,75,45,76,66,45,90,67,88,84,74,45,45,0],"2a26":[103,105,116,104,117,98,46,99,111,109,47,112,118,118,120,0],"2a27":[66,49,46,52,0],"2a28":[86,51,46,48],"2a29":[109,105,97,111,109,105,97,111,99,101,46,99,111,109,0],"2a19":[38],**"2a1f":[4,1],"2a6e":[46,10],"2a6f":[149,9]**,"000102030405060708090a0b0c0d2b12":[0],"1f1f":[85,48,3,21,0,0,40,4,169,124,49,56,60,0,0,0]}}

Thanks a lot!

pvvx, 2021-04-15:
Battery Service UUID is 0x180F and the battery level characteristic UUID is 0x2A19.

closed 2021-04-15 by pvvx

agoralive, 2021-04-22:
I am aware that this is closed, but could somebody explain how this:

**2a1f":[4,1],"2a6e":[46,10],"2a6f":[149,9]**

Amounts to anything worth reading? What is 2a1f (temp???), and is 4,1 then read as 4.1 degrees C?

pvvx, 2021-04-22:
2A1F - **sint16** temperature in 0.1 °C
2A6E - **sint16** temperature in 0.01 °C

**sint16**  - Two bytes

« 2a1f »: [4,1],« 2a6e »: [46,10],

« 2a1f »:  (4+1*256) * 0.1 = 26
« 2a6e »:  (46+10*256) * 0.01 = 26.06