pvvx/ATC_MiThermometer #96: show only external data

bonuzzz, 2021-05-08:
I use sketch to show external data (from mqtt in my case) on lcd, but thermometer has two cycles: 1. it shows my data / 2. it shows data from internal sensor.
Is it possible to disable showing internal data and use the thermometer only like a display?

pvvx, 2021-05-09:
[Primary Service UUID 0x1F10, Characteristic UUID 0x1F1F, command 0x60.](https://github.com/pvvx/ATC_MiThermometer#primary-service-uuid-0x1f10-characteristic-uuid-0x1f1f)

**Xiaomi LYWSD03MMC LCD:**
Packet: 
Set buffer bits LCD: 0x60, RAW_data_LCD[1..6] -> 0x60, RAW_data_LCD[6]
Get buffer bits LCD: 0x60 -> 0x60, RAW_data_LCD[6]

Sample LYWSD03MMC show clock: 
https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L1227

Bitmap buffer LYWSD03MMC LCD:
https://github.com/pvvx/ATC_MiThermometer/tree/master/BoardPinout  -> Mi_LCD_Segments.jpeg

bonuzzz, 2021-05-09:
when I write info to lcd buffer, ext.data stop working. Is it ok?

pvvx, 2021-05-09:
Close Connect -> Restore show LCD.

[UUID первичной службы 0x1F10, характерный UUID 0x1F1F, команда 0x61.](https://github.com/pvvx/ATC_MiThermometer#primary-service-uuid-0x1f10-characteristic-uuid-0x1f1f)
[Restore show LCD: 0x61, 0x00](https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L1256)

closed 2021-05-13 by bonuzzz