pvvx/ATC_MiThermometer #98: not directly project-related problem, , but when try to activating the UART

p-f-w, 2021-05-13:
Hello,

this is not an indication of an error on the original hardware.

However, I would like to use your sources for a bluetooth dongle with the same processor.
There is however on PC2 / PC3 TX and RX from UART. With the sources of AT1441 it was no problem to activate the UART.
Since you have already made extensive improvements, my concern would be to find out from you whether you know what the problem could be. TX works, but RX has no function.
I didn't test non-DMA transfer because I want to use DMA.

Have you changed something in cstartup_825x.s so that the UART cannot be activated? 

however, I also have a mistake here:
"/bin/sh: /cygdrive/c/Users/xxx/AppData/Local/Microsoft/WindowsApps/python3: Permission denied", which I tried to avoid with manual calls, I don't know whether that alone is enough. file is created, runs too, but I'm not sure that the rabbit is not buried there, winpty doesn't help here under Windows 10

pvvx, 2021-05-14:
Сlk disabled for all unused controllers.
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/main.c#L161

p-f-w, 2021-05-16:
Thanks, I had already tried my hand at this point. So my question is whether it might be due to cstartup. I tried again yesterday without success. The port itself still works when I import the modified firmware from ATC1441.

pvvx, 2021-05-18:
> Thanks, I had already tried my hand at this point. So my question is whether it might be due to cstartup. 

No, it doesn't depend on cstartup.s.
> I tried again yesterday without success. The port itself still works when I import the modified firmware from ATC1441.

In BLE, the SoC is constantly put into deep-sleep mode. Everything except the timer and IRQ via GPIO is disabled - power off.
For the UART to work, the SoC needs constant work, except the CPU. This is SUSPEND mode or no transition to any sleep mode.

p-f-w, 2021-05-18:
thanks, that was probably the puzzle solution, before I already had it in the app.c replaced

bls_pm_setSuspendMask (SUSPEND_ADV | DEEPSLEEP_RETENTION_ADV | SUSPEND_CONN | DEEPSLEEP_RETENTION_CONN);
bls_pm_setSuspendMask (SUSPEND_ADV | DEEPSLEEP_RETENTION_ADV | SUSPEND_CONN | DEEPSLEEP_RETENTION_CONN);

by

bls_pm_setSuspendMask (SUSPEND_ADV | SUSPEND_CONN); // SUSPEND_DISABLE
bls_pm_setManualLatency (0);

because it didn't work before, I had to deactivate other parts, but until the end only characters were received sporadically or the program hung up, probably after all it was sent to sleep.

I've now added pin_wake_up for RX and it seems to be working at the moment, now I have to bring some order into it. 

Another Question:
do you know, how to use temperature sensor of TLSR8251? ("The TLSR8251 integrates a temperature sensor and it’s used in combination with the SAR ADC to detect real-time temperature.") Maybe looks like measurement of battery level. But here it hangs on 
"while (!adc_dat_buf[i])"

pvvx, 2021-05-19:
> Another Question:
> do you know, how to use temperature sensor of TLSR8251? ("The TLSR8251 integrates a temperature sensor and it’s used in combination with the SAR ADC to detect real-time temperature.") Maybe looks like measurement of battery level. But here it hangs on
> "while (!adc_dat_buf[i])"

The built-in temperature sensor operates on the Differential ADC inputs. Before measurement, you need to enable power on this temperature sensor in analog registers.
Measurements from this sensor are terrible, and it consumes a lot - it makes sense to use.

pvvx, 2021-05-19:
```
unsigned char     adc_hw_initialized = 0;
volatile unsigned int adc_dat_buf[8] = {1,2,3,4,5,6,7,8}; //size must 16 byte aligned(16/32/64...)

_attribute_ram_code_ void adc_temp_init(void)
{
	adc_set_sample_clk(5);
	adc_set_left_right_gain_bias(GAIN_STAGE_BIAS_PER100, GAIN_STAGE_BIAS_PER100);
	adc_set_chn_enable_and_max_state_cnt(ADC_MISC_CHN, 2);
	adc_set_state_length(240, 0, 10);
	analog_write (anareg_adc_res_m, RES14 | FLD_ADC_EN_DIFF_CHN_M);
//	adc_set_ain_chn_misc(B5P, GND);
//	adc_set_ain_channel_differential_mode(ADC_MISC_CHN, TEMSENSORP, TEMSENSORN);
	adc_set_ain_chn_misc(TEMSENSORP,TEMSENSORN);
	adc_set_input_mode_chn_misc(DIFFERENTIAL_MODE);
	adc_set_ref_voltage(ADC_MISC_CHN, ADC_VREF_1P2V);
	adc_set_tsample_cycle_chn_misc(SAMPLING_CYCLES_6);
	adc_set_ain_pre_scaler(ADC_PRESCALER_1F8);
	adc_power_on_sar_adc(1);
}

_attribute_ram_code_ int get_adc_16bits(void) // run time: first (+adc_temp_init()) 383 us, retry 237 us
{
	int i;

	adc_power_on_sar_adc(1);	// + 0.4 mA
	analog_write(0x07, analog_read(0x07) & (~ BIT(4))); // power up pd_temp_sensor_3V + 0.55 mA

	if(!adc_hw_initialized){
		adc_hw_initialized = 1;
		adc_temp_init();
	}

	adc_reset_adc_module();
	adc_config_misc_channel_buf((unsigned short*)adc_dat_buf, sizeof(adc_dat_buf)); // 8<<4
//	sleep_us(25);
	dfifo_enable_dfifo2();
#if 0 // 0.7 uA
	cpu_stall_wakeup_by_timer0(240*CLOCK_SYS_CLOCK_1US);
#else
	while(reg_dfifo2_wptr < sizeof(adc_dat_buf)>>1);
	while(reg_dfifo2_wptr != 0);
	while(reg_dfifo2_wptr < sizeof(adc_dat_buf)>>1);
	while(reg_dfifo2_wptr != 0);
#endif
	dfifo_disable_dfifo2();
	int adc_result = 0;
	// 14 bit *8
	for(i=0;i<(sizeof(adc_dat_buf)>>2);i++){
		adc_result += (short)adc_dat_buf[i];
	}

	analog_write(0x07, analog_read(0x07) | BIT(4)); // power down pd_temp_sensor_3V
	adc_power_on_sar_adc(0);

	return adc_result>>1;
}

```

p-f-w, 2021-05-21:
thanks, goes so far, only the measurement doesn't seem to really or hardly react to the temperature so far

although that's only a minor problem

I have not yet managed to deliver the data via ble, I'll probably have to make a few changes that data will arrive, unfortunately in the applications it is the same as in the examples with "SPP", different service id's must use for in and out

closed 2021-05-26 by pvvx