pvvx/ATC_MiThermometer #99: Reset device

WildeRNS, 2021-05-23:
How to reset device (LYWSD03MMC)? I set PIN and it's now not working (can't connect - wrong pin) and now I want to totally reset device.

pvvx, 2021-05-24:
What does not work? 
The PIN code can be reset only by the programmer.

https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
“Erase All Flash” and re-record the program.

WildeRNS, 2021-05-24:
> What does not work?
> Enter the set PIN-code and after connecting, reset it by reassigning to "000000".
> 
> A forgotten PIN can only be reset by a programmer.
> https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
> “Erase All Flash” and re-record the program.

Thanks, reflashed and everything works.
On more question: I see I can adjust time clock delta, but don't understand in what units it? Can you describe principal of delta use? Thanks.

pvvx, 2021-05-24:
There is no clock in the chip. Xaiomi regretted solder quartz for clock (32 kHz). The internal RC oscillator is very unstable. Depends on temperature and varies from chip to chip.
Delta value in 1 / 16th of a microsecond. It is entered as an amendment for the account of one seconds.

closed 2021-05-26 by pvvx