pvvx/ATC_MiThermometer #102: blank screen on LYWSD03MMC

hassidan, 2021-05-31:
funny thing...
accidently i flashed on my LYWSD03MMC - CGG1_v31.bin.
and now the screen went blank not showing anything... 
i can still find it in the flasher with a new name ("Qingping Temp & RH M").
but cant reflash it to stock firmware. 
any help? 
here is the log while trying to connect and flash (it gets stuck on activation):
`2:13:09 AM: Searching for devices
2:13:14 AM: Connecting to: Qingping Temp & RH M
2:13:24 AM: Detected Telink OTA service
2:13:24 AM: Connected
2:13:32 AM: Activating now, please wait...
2:26:46 AM: Disconnected.
2:26:46 AM: Searching for devices
2:26:51 AM: Connecting to: Qingping Temp & RH M
2:26:59 AM: NetworkError: Connection failed for unknown reason.
2:26:59 AM: Reconnect 1 from 5
2:27:04 AM: Detected Telink OTA service
2:27:04 AM: Connected
2:27:32 AM: File size: 85924 bytes
2:27:32 AM: Count: 5371
2:27:34 AM: Start DFU
2:27:34 AM: Please do the Activation first
2:36:22 AM: Activating now, please wait...`

pvvx, 2021-06-04:
> accidently i flashed on my LYWSD03MMC - CGG1_v31.bin.
> i can still find it in the flasher with a new name ("Qingping Temp & RH M").

- "CGG1_v3x.bin" uses the BLE name "CGG_xxxxxx".
- The original firmware "Original_OTA_CGG1_v1.0.1_0093.bin" uses the name "Qingping Temp & RH M".

To reflash the original version of CGG1, you need to press the button on its case before connecting...

The key on CGG1 is mapped to GPIO_PC4.
LYWSD03MMC GPIO_PC4 on the board is marked as "P9" ...

It will probably be easier to reflash via a USB-UART adapter: 
https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version
https://github.com/pvvx/ATC_MiThermometer/tree/master/BoardPinout

xfactory9025, 2021-11-24:
The screen seems blank after I updated it too. does not open.

closed 2022-05-01 by pvvx