pvvx/ATC_MiThermometer #104: Configuration Screen only shown for very short time

Huebenthal, 2021-06-23:
I flashed it successfully and try to change the configuration. After connection to the device in the Telink Flasher the configuration screen appears but disappears after 2-3s again and the device is disconnected. Setting any configuration is not possible in this short time.
Chrome browser in Android 11

rocket59, 2021-06-23:
Have the same issue (3.2 firmware) on an old phone (running android 6.01). The device disconnects within a minute of connection and it also takes a few tries to connect/reconnect.

chris4git, 2021-06-27:
I have the same problem : fresh flash of 3.2 from original firmware -> connect -> the form is displayed -> 10s-15s later, auto-disconnect.
A connection on a previously upgraded from 2.x to 3.2 is stable.
Chrome on Android 10

chris4git, 2021-06-30:
Update : I removed and replaced the battery and could change the configuration. So probably a bad contact...

tablatronix, 2021-08-08:
Same, had to reboot it to get it to work again. Odd.

kdre, 2021-08-18:
I have the same problem. A reboot (pulling and reinserting the battery) did not help.
Chrome on Linux: No config options visible at all.
Chrome on Android: Config option visible for short time and then they disappear.

MiguelCosta, 2021-09-09:
I also have issues after update to firmware 3.2

When I try to connect, the device disconnects after a few seconds 😢 

Chrome on Windows, Edge on Windows and Chrome on Android have the same issue.

[image]

closed 2023-02-18 by pvvx