pvvx/ATC_MiThermometer #105: Attaching a DS18B20 probe to replace the sensor

Sukalo0290, 2021-06-23:
I plan on adding one of these water proof probes to replace the sensor but i'm wondering where the data wire would be attached.

[image]

I plan on editing this file 
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/sensor.c

Where would i attach the data wire? I have seen the diagram but where would the data cable go?

Red connects to 3-5V - already know
Blue/Black connects to ground - already know
Yellow/White is data - not sure

[image]

patrickdk77, 2021-11-06:
You cannot simply replace the sensor.
You also have to replace the sensor drive, the display and everything else that uses i2c.

You could add that sensor to any available gpio pin, but you will have to write and add a driver for it also.

jddonovan, 2021-11-07:
> You cannot simply replace the sensor. You also have to replace the sensor drive, the display and everything else that uses i2c.
> 
> You could add that sensor to any available gpio pin, but you will have to write and add a driver for it also.

I'm not sure what you mean? There is no sensor drive -> i2c requires that the attached sensors has a built in microchip for i2c communication, so the SHTV3 is one single package. Devices using i2c can be connected in parallel, so no issue with that, apart from possible address conflicts and power draw issues. You can add and remove sensors from the bus no problem.

The driver is essentially just a string of bytes you send and wait for a response. The bigger issue is how to handle the deep sleep timing and how to incorporate the data in the BLE advertisement package including the code to read the advertisement. Easiest is just to pick a sensor which responds at the same time or quicker than SHTV3 and replace e.g. battery millivolt readings with the new sensor readings.

patrickdk77, 2021-11-07:
True, but a DS18B20 is not i2c, it's 1wire. Did you bother to read what this bug report was about?

If you know what i2c is, then you would know that requires 4 wires, and the DS18B20 only has 3. A brand new driver is needed.

jddonovan, 2021-11-07:
I addressed my reply to everyone stumbling here wanting to attach an external sensor, as stated in my first paragraph. My final paragraph addressed the issue at hand.

The sensor suggested might not be appropriate for a battery operated solution; sink currents are spec'ed at 4mA due to parasitic nature. One can of course cut off the power to the sensor when idle, but recovery time might be an issue. Also one needs to consider the temperature conversion seems to take hundreds of milliseconds. This is based just on a quick glance, but I'd recommend another sensor, possibly an i2c. Personally I'd just get the same Sensirion sensor with a different address, then dip sensor&wires in epoxy or something to make it waterproof. Humidity measurements obviously won't be correct, but it would be relatively easy solution.

nagyrobi, 2022-03-19:
DS18B20 is a poor sensor compared to the built-in STHx3. DS18B20 has 0.5 degrees precision at most (originals! chinese clones have 0.8-1.5 degrees in error!). Also absolute values need offsets to be calibrated.
Sensirion has 0.2 degrees precision and it's also much closer to reality in absolute measurements too.
Forget the DS18B20, you already have the best possible sensor in LYWSD03MMC.

pvvx, 2023-03-28:
The DS18B20 cannot be powered by CR2032. Does not fit into the permissible operating voltage. The voltage of the CR2032 at load is less than 2.9 V 90% of the operating time. Thus, the DS18B20 consumes a lot of energy during measurement (more than 1 mA 950 ms) and at least 5 µA continuously only on the signal wire.
In addition, according to the documentation, the self-heating of the DS18B20 when measuring the current consumed by it and taking into account the dispersion of the TO-92 housing is 5V * 0.0015A = 0.0075 W, which gives an increase of +2.5 C.
Who needs such an outdated and inaccurate thermometer?

closed 2023-03-28 by pvvx