pvvx/ATC_MiThermometer #108: Possible instant save to memory in case of fast temp or humi changes

zian31, 2021-06-26:
In the atc1441 firmware, there is one special functionnality :
"Temp or Humi instant advertising
When the temp or Humidity changes too fast between the main loop (5 seconds interval), the Advertising will be instant for that one.
byte0 0xFC = temp_alarm_point // value divided by 10 for temp in °C
byte0 0xFD = humi_alarm_point"

Maybe it could be fine to get this kind of functionnality concerning an instant memory save in case of fast temp or humi changes (instead of an instant advertising with the atc1441 firmware).

I'd like to put an ATC_Thermometer in my fridge for one year with one minute interval between memory saves, and be able to also see the moments when I open the door of the fridge  : I think there is fast change of temp in this case which is impossible to track with the actual firmware and one minute interval : moreover if I choose to use the very nice average system -> so the fast changes would have a MSB at 1 for example to make a difference with the average values in memory ?

Thanks for reading.

pvvx, 2021-06-27:
**ATC1441 firmware:**
Advertising interval = fixed  3000 * 0.625 ms = 1875 ms = 1.875 sec
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app_config.h#L9
Always transmitted with a pause of 1.875 seconds + random (<50 ms)!

Minimum step of updating data for transmission = minimum 5000*1.5 ms = 7500 ms = 7.5 sec.
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app.c#L125

Measure interval = minimum 10 sec.
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app.c#L77
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app.c#L85

---

In my version, all parameters are configurable.
The minimum step of polling and data transmission from the sensor = 62.5 ms

zian31, 2021-06-28:
Thanks victor,
I mean this kind of configuration in your version :
- Example parameters used in your version : Save to Flash memory date+temp+humi+batt each 10 minutes, with an average of 60 values, so one mesure of temp+humi+batt is done each 10s : acceptable for consumption
- New possible feature : If temp or humi change more than 1° or 1% between mesure interval (here 10s), new save of date+temp+humi+batt in the Flash memory without averaging => it's exeptionnal saves due to fast changes (so also acceptable for consumption), to recognize these kind of changes, MSB of batt value can be set to 1 for example ?

pvvx, 2023-02-18:
Periodic ads are not served in Home Assistant and other apps. Linux has just started to support the Bluetooth v5.0 standard released in 2016, but most top-level programs don't work with Bluetooth v5.0.

closed 2023-02-18 by pvvx