pvvx/ATC_MiThermometer #109: Complete deactivation of smiley on MHO-C401 not possible

Flugente-19, 2021-06-29:
Today I flashed the FW v3.1 on two devices LYWSD03MMC and MHO-C401. There is a small issue with the MHO-C401: It is not possible to completely deactivate the smiley. When I select the blank smiley, the two brackets are still visible. Could you please check it?

[image]

MacSass, 2021-06-29:
Hi,
yes - that is because how the brackets are linked to the . by hardware - thus they are not individually adressable per my understaning. See closed issue and explanation here: [https://github.com/pvvx/ATC_MiThermometer/issues/63](url)

Flugente-19, 2021-06-29:
Okay. Thank you for the fast response.

closed 2021-06-29 by Flugente-19