pvvx/ATC_MiThermometer #110: Cannot download the data with GraphMemo.html

tglaria, 2021-07-14:
Following your guide I flashed the device with after some issues (it was dificult to connect to the thermometer).
I left the default configuration but changed the name.
Then I try to download the stored data from the device from the webpage GraphMemo.html but I can't.
I click on the connect button and select the flashed device with the custom name, but can't.

Sometimes says: Says Connecting to GATT server
and then: NetworkError: Connection failed for unknown reason.

Another time it stuck on: GATT server connected, getting service...

And aanother time it stays stuck at: Send command get-memo (0..19632 samples)...

I get the same problem on Edge and Chrome browsers.

Might I be missing something?

traeu, 2021-08-24:
I have the same problem. I think the connect-functionality of GraphMemo.html is somehow broken... 
I found a workaround: Connect first with TelinkMiFlasher.html. Connecting to TelinkMiFlasher.html works for me without problems. Then open GraphMemo.html in a new tab and click connect. The symbol of your device in the list of scanned devices should indicate now that you are already connected. Now you should be able to connect (at least thats how it works for me)

tkerby, 2022-01-03:
Same here. also noticing it only seems to manage to download about 60 readings

closed 2022-05-01 by pvvx