pvvx/ATC_MiThermometer #112: MCCGQ02HL activation

lolax80, 2021-07-30:
Hi,

can you add the support to activate the MCCGQ02HL Door Sensor2?

Thanks

pvvx, 2023-12-23:
What chips are used in MCCGQ02HL?

closed 2023-12-23 by pvvx

Ircama, 2023-12-26:
The MCCGQ02HL (Xiaomi Door and Window Sensor 2) presents itself with the "standard demo" label and it does not include a Telink chip.

Someone tore down the device to show its internals.

[image].

It should use a Silicon Labs chip, perhaps the [EFR32BG21A010F512IM32-B](https://www.silabs.com/documents/public/data-sheets/efr32bg21-datasheet.pdf), Bluetooth 5.1 (A010HI means MCU A0, Radio Feature Codes 10, H = 512k, I = -40 to 125 °C).

The [TelinkFlasher from pvvx](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) is not able to return its key:

```
10:50:33 AM: Searching for devices
11:31:18 AM: Connecting to: standard demo
11:31:25 AM: Not found Telink OTA service!
```

Also, the [TelinkFlasher fromatc1441](https://atc1441.github.io/Temp_universal_mi_activate.html) shows "Connected", but does not activate the key:

```
11:34:49: Connected
11:35:08: Activating now, please wait...
```

The only way to get the Bind Key was by using method [3. MiHome mod (Android only)](https://custom-components.github.io/ble_monitor/faq#encryption-keys) and then with the token extractor script "5. get_beacon_key python script".

The configuration of the device can be performed via Xiaomi [MiHome mod](https://www.vevs.me/2017/11/mi-home.html):

- Ambient light detection enablement, thresholds and detection time
- Door open timeout enablement and timer value
- Firmware update

Its advertising traffic can be fully decoded with the new version of [atc_mi_interface](https://github.com/Ircama/ATC_MiThermometer/tree/ircama-SJWS01LM/python-interface), not yet merged because still under development at the time of writing. When using the GUI, it relies on a new version of the [construct-gallery](https://github.com/Ircama/construct-gallery) library, again not yet published to PyPI because under development.

Its MAC address is from "Tianjin Hualai Tech Co, Ltd" (E4:AA:EC:...).

The generated events are the following:

- XIAOMI_DATA_ID_DoorSensor:
  - XIAOMI_door_closed
  - XIAOMI_door_open
  - XIAOMI_door_not_closed_in_time

- XIAOMI_DATA_ID_LightIntensity
  - light_detected (True or False)

- XIAOMI_DATA_ID_Power
  - battery_level

Perhaps in order to overcome possible collisions, the device communicates through a train of advertisements (e.g., at least 3), which are typically preceded by a XIAOMI_UNBOUND advertisement. Related detection can be done via active or passive advertisement discovery. It is important to configure the host BLE device driver to avoid removing duplicated advertisements (e.g., subsequent data closely coming from the same MAC), otherwise, since they are also very close to each other, the only detected message could be XIAOMI_UNBOUND, or even no data at all sometimes.

Using Python, Windows passive advertisement works with [bleak](https://bleak.readthedocs.io/) using `scanning_mode='passive'`. With Linux, Bluez defaults with a configuration tuned for servers, will battle-tested settings to reduce scanning to the minimum in order to avoid floods of messages to the D-Bus IPC. To configure passive scanning, also disabling the removal of duplicated advertisements, a new version of Bluez is needed (e.g., >= 5.56, like 5.70) with `ExecStart=/usr/libexec/bluetooth/bluetoothd --experimental` in `/lib/systemd/system/bluetooth.service`. Also, [Disable "Filter duplicates" setting for LE Set Scan Enable command](https://stackoverflow.com/questions/55336017/disable-filter-duplicates-setting-for-le-set-scan-enable-command) shall be issued after the Bluetooth discovery is active: `hcitool cmd 0x08 0x000C 0x00 0x00 && sudo -n hcitool cmd 0x08 0x000C 0x01 0x00`. The following example is a working Python program that can print all events.

```python
import subprocess
import asyncio
from bleak import BleakScanner
from functools import partial
from atc_mi_interface import general_format, atc_mi_advertising_format
from construct import EnumIntegerString

from bleak.assigned_numbers import AdvertisementDataType
from bleak.backends.bluezdbus.advertisement_monitor import OrPattern
from bleak.backends.bluezdbus.scanner import BlueZScannerArgs
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData

bindkey = {
    "E4:AA:EC:AA:BB:CC": bytes.fromhex("aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa"),
    # ...
}
[... 46 lines cut ...]
        print('-' * 80)
        count[0] += 1
        if count[0] == MAX_COUNT:
            stop_event.set()

    ble_args = BlueZScannerArgs(
        or_patterns=[
            OrPattern(0, AdvertisementDataType.FLAGS, b"\x06"),
        ],
    )
    async with BleakScanner(
        detection_callback=partial(detection_callback, count),
        scanning_mode="passive", bluez=ble_args
    ) as scanner:
        subprocess.run(
            "which hcitool && sudo -n hcitool cmd 0x08 0x000C 0x00 0x00"
            " && sudo -n hcitool cmd 0x08 0x000C 0x01 0x00",
            shell=True,
            capture_output=True
        )         
        await stop_event.wait()
    print("Stopped")

asyncio.run(ble_coro())
```

pvvx, 2023-12-26:
All Xiaomi transmit data in blocks of 16 advertising events at short intervals. This amount is always accepted by a regular BT adapter.