pvvx/ATC_MiThermometer #113: CGG1 no activation

klim123123, 2021-08-03:
Hello,
I have successfully flashed LYWSD03MMC and MHO-C401. No problems at all.
When i try to flash CGG1 i'm stuck at Activating now, please wait...
I tried two computers with BT dongles, i tried Android phone. Always: Activating now, please wait... and nothing....
I have tried 3x different CGG1. All of them version CGG1-M with qingping logo at the back.
Tried to change batteries.
CGG1 are connected to Mi Home hub.

16:40:00: Searching for devices
16:40:07: Get Advertising MAC (Web Experimental Features!)
16:40:07: Connecting to: Qingping Temp & RH M
16:40:07: MiVer5, DevID: 0x0B48-CGG1_ENCRYPTED, FnCnt: 10, CtrID: 0x5858 Not bound, Standard certification, MAC: 582D3412BBAA, Data encrypted
16:40:11: Detected Telink OTA service
16:40:11: Connected
16:40:18: Activating now, please wait...

klim123123, 2021-08-04:
After 2 days of trying and trying, i found solution.
1. Connect to device using https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
2. Wait Connected message.
2. Click to Activate and you will stuck at "Activating now, please wait..."
3. Hold button at the back of CGG1 for 3 seconds, you will see bluetooth icon blinking on device.
4. Click Reconnect on page.
5. Wait Connected message.
6. Now you will be able to Activate (Click to Activate button)! You can upload custom firmware to your CGG1 device.

pvvx, 2021-08-07:
https://github.com/pvvx/ATC_MiThermometer/issues/41#issuecomment-830570681

closed 2021-08-07 by pvvx