pvvx/ATC_MiThermometer #114: Azarton MHO-C401 Display freez (MHO_C401_v32n.bin)

Rocket200, 2021-08-07:
i flashed this "MHO_C401_v32n.bin" FW but  the Display freeze all other work .
can you fix this ?
i can send you one , or what can i do?

https://www.youtube.com/watch?v=Zy-qYkfq1YQ

BoxedBrain, 2021-09-07:
Having the same error. After flashing the Azarton MHO-C401 the display is frozen/stuck.
I get readings but the display won't update. Also, a reflash to the original/custom firmware doesn't solve it.

Any ideas or updates on this?

Rocket200, 2021-09-18:
@pvvx can you repair / fix the freezed Displays with a new FW?

Rocket200, 2021-09-25:
@pvvx can flash new and repair it?

BoxedBrain, 2021-09-27:
@Rocket200 I tried again to reflash but without success. Could a picture of the PCB help? Maybe it is a newer revision? Different GPIO pin or something?

BoxedBrain, 2021-09-28:
[image][image][image][image][image]

Rocket200, 2021-11-01:
this new CFW fix it ? 
MHO_C401_v33.bin

pvvx, 2021-11-02:
Xiaomi Miaomiaoce MHO-C401 has a different E-ink display and different PCB.

Rocket200, 2021-11-02:
> Xiaomi Miaomiaoce MHO-C401 has a different E-ink display and different PCB.

all working fine … with your FW only the Display is freezing and not show the correct Temp.

pvvx, 2021-11-02:
Xiaomi Miaomiaoce MHO-C401 has a different E-ink display:
[image]

Rocket200, 2021-11-02:
[image]

i have 3 MHO from Xiaomi too
but Display look same style

Rocket200, 2021-11-02:
ok thanks … i see it now

evildad, 2021-11-25:
Well I should have read this first but I've flashed the Firmware to my Azarton-MHO-C401 and have the freeze, too.
The readings work fine but having the Display back would be an enhancement :-)
The Custom-FW doesn't work because we have a different PCB and Display and that's the Problem for the Recovery FW as it's not from the Azarton Device.
@pvvx Is there a way to make a backup from a working Azarton-MHO-C401 and flashing that back to the broken one?

pvvx, 2021-11-25:
Make a Flash backup of a running Azarton-MHO-C401

1. Using [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) download '[BLE-Hacker.bin](https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker/blob/main/BLE-Hacker.bin)'
2. Using [TLSR825xOTA_Hacker.html](https://pvvx.github.io/CGPR1/TLSR825xOTA_Hacker.html), connect to the appeared device "BLE_xxxxxx"
3. Restore original firmware boot: "Change Start Boot" key. (Do it once!)
4. Read All Flash to File: 'Read Full Flash' and 'Save' to File.
5. Close 'TLSR825xOTA_Hacker.html'.

The details: https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker

evildad, 2021-11-26:
Ok sounds like a plan for the weekend :-) Just to be sure... the working one will boot the orginal firmware again after step 5?

leoguiders, 2021-11-26:
Ah great, I installed the custom FW on my Azarton MHO-C401 today and also have the frozen screen.
Fow now I don't have a working one to download the firmware, I'd appreciate if you could upload the original firmware.

pvvx, 2021-11-26:
> Ok sounds like a plan for the weekend :-) Just to be sure... the working one will boot the orginal firmware again after step 5?

Yes. 
The main thing is to read FullFlash and make OTA.bin...

evildad, 2021-11-27:
@pvvx ok it seems like it worked and best of all it's still working :-)
Attached is the FullFlash.bin 
[azarton_mho-c410_ff.bin.gz](https://github.com/pvvx/ATC_MiThermometer/files/7611628/azarton_mho-c410_ff.bin.gz)

Rocket200, 2021-11-27:
soon a new FW for it ?

for repair it and flash back the original FW?

pvvx, 2021-11-27:
https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Azarton_MHO_C401_v1.0.0_0016.bin

evildad, 2021-11-27:
@pvvx perfect... successfully unfreezed it 👍

pvvx, 2021-11-27:
@evildad Thank you for providing Full Flash!
I checked the check-sum and firmware functionality on another module with TLSR825x. But I don’t have Azarton MHO-C401 and so far I can’t make support for new E-Ink.

leoguiders, 2021-11-27:
> I don’t have Azarton MHO-C401 and so far I can’t make support for new E-Ink.

What do you need to implement the new e-ink?

pvvx, 2021-11-27:
> What do you need to implement the new e-ink?

Time and opportunity to buy a "Azarton MHO-C401". They are not in the sale available to me.

leoguiders, 2021-11-28:
I'd be happy to send one to you

leoguiders, 2021-11-29:
Another piece of information:
The FPC cable has HB0168 printed on it, which might or might not lead to the manufacturer of this e-ink display.
[image]

Does anyone know how to remove the display from the Case?

Lightning-, 2022-08-16:
@pvvx is there a chance that you can proceed with this device? i just bought one and like it quite a lot; would love to get this flashed with a custom firmware and get some more of them.  
for me these azarton devices are available on amazon; if this is true for you too i'd be happy to donate one or two to make it possible for you to free them for us! :smiley:

kloemi, 2022-10-09:
For me the display of the azarton looks quite similar to the one of MHO-C201 (https://github.com/jonathonlui/MHO-C201) Has anyone ever tried that code for the azarton?

pvvx, 2022-10-10:
> For me the display of the azarton looks quite similar to the one of MHO-C201 (https://github.com/jonathonlui/MHO-C201) Has anyone ever tried that code for the azarton?

[image]

[Xiaomi Mijia MHO-C401](https://pvvx.github.io/MHO_C401/) -> https://github.com/znanev/MHO-C401
[image]

kloemi, 2022-10-10:
OK, you're right FPC looks same like MHO-CD01. I thought because of the'x' in the batterie symbol.
Azarton:
[image]

MHO-C201:
[image]

evlo, 2022-10-12:
It did not occurred to me that it is not just rebranded xiaomi. Xiaomi BT ones with LCD are the best TH meters I used, I then bought MHO-C201 by accident and then found Azarton on local Amazon.

I guess for now the ones with LCD are a way to go for TH measuring devices and just think of it as displayless device.

kloemi, 2022-10-23:
I have analyzed Azarton display and implemented a version at fork https://github.com/kloemi/ATC_MiThermometer.
There are 3 minor issues
* initial value is shown ~10-12 sec. after startup
* degree sign not show
* battery sign not shown
For me it works ok now, maybe someone else has an idea about battery and degree sign. It seems not to be a part of the display data buffer.

Thanks @pvvx  for your work!

SoulEater45, 2022-11-28:
> I have analyzed Azarton display and implemented a version at fork https://github.com/kloemi/ATC_MiThermometer. 

I tried to flash both [MHO_C401_v37c.bin](https://github.com/kloemi/ATC_MiThermometer/blob/master/MHO_C401_v37c.bin) and [MHO_C401_v38.bin](https://github.com/kloemi/ATC_MiThermometer/blob/master/MHO_C401_v38.bin) on my Azarton and my display still keeps freezing...
Am I doing something wrong? I flashed the original Firmware and then yours, but seems like I am missing something. Sorry if it is a dumb question, but I am a bit new to all of this

kloemi, 2022-11-28:
Hey, as mentioned in my repo description: due to the existing issues of my software I did not insert the binaries. You can build the SW on your own. The binaries inside the repo are the forked ones from victor.

evlo, 2022-12-08:
I guess there is new version of regular MHO-C401 a display freezes on that too now.

_(I returned the Aarton and bought the original) ... turns out that was the mistake :)_

Bascht74, 2022-12-30:
Hi, I got into this problem, too... right now, two of my device displays are not working anymore after flashing 3.8/3.9. 
I am not sure, which firmware is needed to get back to the original. 

@kloemi I think I could live with those issues, because I check the device battery with Home Assistant, and a degree sign I don't need as I know the unit :-) The 12 seconds are fine, as I don't need to change the battery too often. 
So maybe you could add a bin file at your repository (maybe you could add that to an "alpha" directory to sign it "not completely working"? This would help me a lot as the original firmware does measurement-update only every 10 minutes or so and the custom one every 10 seconds. 

If you have second thoughts about that, could you advise me how to compile the firmware myself?
(Maybe anybody else from this entry has already done it and could provide me a bin file?)

@pvvx Could you add a warning to your readme to prevent new people flashing their devices with the custom firmware for the other hardware revision (maybe a link to this issue, e.g: "Azarton devices --> please read https://github.com/pvvx/ATC_MiThermometer/issues/114 ? I think this would help other people to decide and read this issue.

pvvx, 2022-12-31:
> If you have second thoughts about that, could you advise me how to compile the firmware myself? (Maybe anybody else from this entry has already done it and could provide me a bin file?) This would help me a lot as the original firmware does measurement-update only every 10 minutes or so and the custom one every 10 seconds.

This means incorrect initialization of the controller.  And how it will work further and will not lead to the degradation of the E-Ink panel is unknown.
 
> @pvvx Could you add a warning to your readme to prevent new people flashing their devices with the custom firmware for the other hardware revision (maybe a link to this issue, e.g: "Azarton devices --> please read #114 )? I think this would help other people to decide and read this issue.

Supported thermometer models are listed in the repository header.

kloemi, 2023-01-11:
Hey @Bascht74 ,

sorry for late response I have uploaded the binary that is running fine on my  azarton thermometer since 2 month. You can try if that is useful for you.

pvvx, 2023-02-27:
Version 4.2 works on "Azarton"?
MHO_C401N_v42.bin version works on new MHO-C401(2022.11) without "Azarton" inscriptions.

kloemi, 2023-03-11:
Thanks for you great work! MHO_C401N_v42.bin works fine, I think you can close that issue.

closed 2023-03-11 by pvvx