pvvx/ATC_MiThermometer #117: Humidity (LYWSD03MMC)

ACrazyConcept, 2021-08-23:
Hi

Finally got my LYWSD03MMC flashed. Totally cool improvement.

I was just wondering if the humidity sensor is also able to provide the same precision and update frequency as the temperature?

[image]

tetsuo55, 2021-08-23:
Mine are as detailed with humidity as they are with temperature, strange that yours stays like that

giannoug, 2021-08-24:
Same here, values seem to have the same precision:
[image]

Have you noticed that humidity might need offset setting? Not sure if the default configuration reports the correct values.

Can anyone confirm?

ACrazyConcept, 2021-08-24:
I just added the sensor on a new HA and there I get the same level of detail in both humidity and temperature. Still strange though. But it's resolved.

closed 2021-08-24 by ACrazyConcept