pvvx/ATC_MiThermometer #118: Can't connect after updating firmware to 3.2

acronce, 2021-08-30:
I flashed an LYWSD03MMC to v3.2 and after it succeeded I could not reconnect to the device. It doesn't even show up as a bluetooth device any more in the Windows -> Settings -> Bluetooth.

The device itself seems happy. It's displaying the temp and humidity like before. 

I power cycled it by removing the battery and putting it back in. It still displays the temp, but now longer appears as a Bluetooth device.

Any suggestions?

pvvx, 2021-08-30:
https://github.com/pvvx/ATC_MiThermometer/issues/100

acronce, 2021-09-01:
That seems to be a different problem. In my case after flashing I cannot see the device in the Telink Flasher. It basically disappeared after flashing and will not reconnect.

jddonovan, 2021-09-04:
I have zero issues with my Windows computer with a relatively new bluetooth adapter, supporting BT5.0. I also have an atomic pi with BT 4.0 using bluetooth2mqtt and I couldn't get any sense on why it wouldn't connect or read the mi-thermometer whereas e.g. my smart band was detected ok.

Someone in the review section mentioned that they could not get this to work on BT 4.0 adapter, but using a BT 4.2 it worked fine. Could this be the reason? The sensor requiring version 4.2 minimum for successful broadcast after flashing?

MiguelCosta, 2021-09-09:
I also have issues after update to firmware 3.2

In my case, after flashing I can't configure the settings. When I try to connect, the device disconnects after a few seconds 😢 

[image]

acronce, 2021-09-26:
I moved to a different computer (a Mac running Chrome) and was able to connect to the thermometer. This allowed me to configure the device with no further issues.

closed 2021-09-26 by acronce