pvvx/ATC_MiThermometer #121: increase transmission power?!

50494554524F, 2021-09-12:
reading chips specs i was wondering if it possible to increase the transmission power more than +3db...

thansk

pvvx, 2021-09-18:
This option is not available for the current LYWSD03MMC schema. The chip can, but the printed circuit board and other design do not allow an increase of more than +0.3 dB. At a higher power in the chip, the power supply of the RF part is switched ... The battery current is critically lacking. As a result, the radiated power drops very much ...

50494554524F, 2021-12-04:
sorry, me again...

LYWSD03MMC has a really wide range but i'm still fighting with reinforced concrete in a couple of points and i need a little more power...

can you / anyone suggest a bluetooth repeater or an easy and cheap to build diy project for a bluetooth repeater?!
it has to be cheap otherwise using another board will be preferable...

thanks

pvvx, 2021-12-05:
As of today, there is only a repeater https://github.com/pvvx/AdScanerTrg
Uses connection mode ...

Replacing the CR2032 with a more powerful battery allows for higher TX RF power to +10 dB. A pulse current of 30 mA is required at 3.3 V.
In TelinkMiFlasher.html, [change line](https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L28): 
```` const MAX_RF_TX_Power = false; ````
to 
```` const MAX_RF_TX_Power = true; ````
[image]

50494554524F, 2021-12-23:
much much better using +10 db, have to find a optimal position but it is quite usable now (2xAA batteries)... 

now i'm going to test your adscannertrg as repeater (great idea), a couple of question before proceed:

- is there any possibility to brick the thermometer?! sorry, what version have i to install for LYWSD03MMC B1.4?!

- can i put back on your "regular" firmware when done with the thermometer used for tests?!

thanks

pvvx, 2021-12-24:
OTA is standard there.
For the LYWSD03MMC PCB, the AdScanerTrg version has not been completed yet. It is necessary to insert the indication on the LCD.

Viper780, 2021-12-30:
What's the max. Voltage the LYWSD03MMC can handle?

gumby2, 2022-11-30:
> What's the max. Voltage the LYWSD03MMC can handle?

I connected a lipo 720mah 1s battery and it works fine in my freezer, measured at 4.2v when fully charged.  I woukd like to increase tx power but the config tool doesnt work for me when i copy all the files to my local server.  server isnt set up for https so that might be the issue.

pvvx, 2022-11-30:
> What's the max. Voltage the LYWSD03MMC can handle?

3.6V

gumby2, 2022-11-30:
> > What's the max. Voltage the LYWSD03MMC can handle?
> 
> 3.6V

Yes, that is the official spec.  One issue I had is using the 3.9 SW with type MIJIA , config app reports 4.091v while the voltage reported in homeassistant is 2.36v.  you would think it would be reporting 3.1v since that is the dcdc internal output.  time will tell if this voltage level will harm the chip though for now I've seen no issues, though the low temperature of the freezer it's in is lowering it's voltage somewhat.

gumby2, 2022-12-01:
Managed to point chrome browser running on firehd10 tablet  to local folder containing the config app files and increased tx power to max and am observing incredible range.  I placed the sensor at furthest distance in house and am still getting -63dbm when scanned from my phone.  My original purpose was to use the device as a tracker I think this will do nicely.

Great work Victor!  I appreciate all your efforts.

One suggestion on the new firmware voltage sensor, why not make it the same as vbat as seen on the config app?

Scott

gumby2, 2022-12-01:
Yes, that is the easy way and I tried it first, but my old hp8000 doesn't have ble and my pi4 running raspbian buster wouldn't connect even after pairing seemed to succeed.  Chrome on the android platform doesn't have devtools so that was out of the question.  Just glad to get it working without having to buy hardware.

closed 2023-02-18 by pvvx