pvvx/ATC_MiThermometer #122: Where is the reed switch?

Lenbok, 2021-09-13:
I have flashed a LYWSD03MMC with 3.2 firmware and wish to use the reed functionality to monitor the status of a nearby door.

The instructions say the reed switch should be connected on "GPIO PA6 (label on the P10 pin)", however the pinout on https://github.com/pvvx/ATC_MiThermometer/blob/master/BoardPinout/TabPins.gif lists PA6 as being with the mark "P8", and that "P10" is PB6. Which is the correct solder point?

I did try soldering a reed switch to GND and "P10", and when I activate the reed switch the LCD blanks for a few seconds then returns. When I deactivate the reed switch, the LCD blanks again and does not wake up. It seems that either "P10" is the wrong point, or there is a bug in the firmware.

Please clarify!

pvvx, 2021-09-18:
[LYWSD03MMC -> GPIO_PA6 (pin 3 TLSR8251F512ET24) -> mark "P8"](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app_config.h#L203)
CGG1 -> GPIO_PD3
MHO_C401 - none (no free pins)

Lenbok, 2021-09-26:
Thanks @pvvx 

So, I have soldered a reed switch to GND and "P8".

I understand that I need to set the firmware Advertising type to "Custom", and also check the "AdFlags" option in order for reed switch status to be transmitted. When I connect via Telink Flasher, I get "flg: 0x0F", which indicates reed is closed since first bit is set. However, if I allow the reed to open, I do not see any change in the "flg" value. "Count" is increasing, so new records have been received. Is this how I should be checking the reed switch status, or am I missing something else?

My config reports "Custom config: [1, 16, 0, 0, 40, 4, 169, 124, 49, 56, 60]"

pvvx, 2021-09-26:
The existing problem:
The GPIO is very sensitive. The internal pull-up resistor has high resistance. Short-term switching on of the internal pull-up resistance before measurement may not have time to charge the capacitances of the wires or overpower the leaks. Continuous activation of the internal pull-up resistor provides significant additional current to the battery when the contact is closed ...
In the firmware, the interval of preliminary activation of the internal pull-up resistance is not optimally debugged ...

tb74, 2021-10-09:
Currently with firmware 3.2 the reed switch bit just follows the state of GPIO_TRG output. The reason is a small mistake in the test_trg_input function (which I think should rather be named "test_rds_input"), that causes it to read the state of the wrong pin.

The reed switch feature will work if you change line 42 of trigger.h: https://github.com/pvvx/ATC_MiThermometer/blob/d1552569a7cfecf0234a62b9d6a81b78252c0d26/src/trigger.h#L42
Replace `GPIO_TRG` with `GPIO_RDS`:
`trg.flg.rds_input = ((BM_IS_SET(reg_gpio_in(GPIO_RDS), GPIO_RDS & 0xff))? 1 : 0);`

Lenbok, 2021-10-10:
Sounds promising! @pvvx can you please make these changes and compile a new build?

tb74, 2021-10-11:
@Lenbok in the meantime you could use the firmware that I've compiled to test the change I suggested. It's essentially the same as ATC_Thermometer32z.bin, the only change is the one described above.
[ATC_Thermometer32z+rds_fix.zip](https://github.com/pvvx/ATC_MiThermometer/files/7322221/ATC_Thermometer32z%2Brds_fix.zip)

closed 2021-12-12 by pvvx