pvvx/ATC_MiThermometer #123: NotFoundError: No Services matching UUID 00010203... What am I doing wrong?

Cr4z33, 2021-09-15:
I am trying to custom flash the first of my four Xiaomi CGG1, but I keep getting this.

Please how can I sort it out?

`12:43:59: Searching for devices
12:44:03: Connecting to: ClearGrass Temp & RH
12:44:10: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:44:10: Reconnect 1 from 5
12:44:10: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:44:10: Reconnect 2 from 5
12:44:10: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:44:10: Reconnect 3 from 5
12:44:10: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:44:10: Reconnect 4 from 5
12:44:10: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:44:10: Reconnect 5 from 5
12:44:10: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:44:10: Something went wrong, to many reconnect's`

pvvx, 2021-09-18:
Only CGG1-M is supported.
https://github.com/pvvx/ATC_MiThermometer/issues/41#issuecomment-812803456

Cr4z33, 2021-09-19:
OK thanks, then I'll move to some Zigbee ones.

Too much battery drain with the default firmware. 👎🏽

closed 2021-12-12 by pvvx