pvvx/ATC_MiThermometer #124: Set much longer advertising intervals/turn off advertising altogether

Goldrush12, 2021-09-16:
I would like to use much higher advertising intervals, say every 10 minutes or 15 minutes. 

Additionally, it would be quite handy, if advertising could be turned off altogether for use cases where only temperature tracking to flash or the clock mode is desired.

Any way to make this possible?

pvvx, 2021-09-18:
It is possible, but in a separate project ...
The BLE standard does not support advertising intervals longer than 10 seconds.
It will not be possible to connect to the device. Typical external BLE equipment performs poorly already at intervals of more than 2.5 seconds.

Goldrush12, 2021-09-19:
I'm only using the advertising mode, so I assume it would still work fine with longer advertising intervals? I could manually reset the device if I wanted to change settings, right?

Any chance you could implement that?

pvvx, 2023-02-18:
Periodic ads are not served in Home Assistant and other apps. Linux has just started to support the Bluetooth v5.0 standard released in 2016, but most top-level programs don't work with Bluetooth v5.0.

closed 2023-02-18 by pvvx

Stephan3, 2023-09-21:
> It is possible, but in a separate project ... The BLE standard does not support advertising intervals longer than 10 seconds. It will not be possible to connect to the device. Typical external BLE equipment performs poorly already at intervals of more than 2.5 seconds.

Hello @pvvx . 

Found this using search function, as i am was after the same thing. Currently i use 12.5s advertising interval. Form temp and Humidity 30s+ would be still more than sufficent. I dont know if i understand the connection mechanics right.
So if a longer advertising/messurement interval would result in longer battery life, why not?
I transfer the data using few tasmota devices here and there to my iobroker. Works great.

If this has been answered elsewere, sorry to spam you here. 

--
Stephan

pvvx, 2023-09-21:
> Currently i use 12.5s advertising interval. 

The official BLE standard limits the advertising interval to 10 seconds.
Recommended, according to the results of implementation - up to 1.6 seconds. The reasons are negligent software and installations in the Linux kernel (introduced sabotage from Intel specialists, plus an incorrectly selected algorithm for working in the driver - blocking the driver for the scanning period), old BT adapters, restrictions in recommendations from Apple contrary to Bluetooth SIG standards, ...

Currently, in some adapters, the embedded software is limited to a scan period of up to 10 seconds. It simply does not accept or does not connect, because a timeout of less than 10 seconds is built in and requires 2 receptions for this period.

>The devices with a button advertise much more often when button is pushed. So connection could still work.

Yes. All thermometers with a button and alternative firmware have the ability to enable the "Connect" function. This function shortens the advertising interval by 1..1.6 seconds so that you can connect. The operation time of the function after pressing the button is about 80 seconds.

Stephan3, 2023-09-21:
Thanks for your answer. I thought "just make it 30 seconds and tasmota will catch it". Obviously that is not possible.

pvvx, 2023-09-21:
Support for the BLE 5.0 standard will become possible when the BT drivers are completely rewritten in Linux. This cannot be done using an external API. The adapters themselves have nothing to do with it.