pvvx/ATC_MiThermometer #125: Cannot seem to get it to work on version B1.9

bbosilje, 2021-09-28:
When I try to flash it with the latest firmware in the repo, the display remains "frozen" with the last values it displayed with the stock firmware.

When I flash back to the original firmware on the web page, the display comes back with all the segments lit up (ie: 1888.8 E, 88 %, Bluetooth icon, battery icon and full face).

I am still able to communicate with the device through the flashing webpage, but nothing I do seems to change the display.

Am I doing something wrong or does this revision of the unit (B1.9) not work with the existing firmwares in this repo?

Toaster232, 2021-09-28:
I have the same problem with 2 unit's. Both have CMIIT ID: 2019DP8115 (On the back of the carton box)
After installing de original software back on the unit. The display shows 88% and LOW battery icon and says the temp is 27c when looking at the ESPHome log.

vanMacG, 2021-09-29:
My first thermometer works fine, so I ordered some more. With  he new ones I have the same problem. After flashing a custom firmware (31, 32n, 32z) the display freezes. When putting the battery out and back in the display stays empty. But I'm able to connect with the webupdater. After flashing the original firmware back, the display is like the picture in first post.

Here are pictures of the packaging and the PCB:
[image]
[image]
[image]

vanMacG, 2021-09-30:
Besides the display problem, the measurement also doesn't work. After sucessfully connecting with the webupdater, at the top of the page, it says: "Temp/Humi: waiting notify for data after connecting".
When using GraphMemo after some time: "load 0 samples."

Maybe @atc1441 knows the new ones and can help...?

dvonessen, 2021-10-02:
I also have a similar problem. Except, using the original FW doesn't bring back the LCD. 
Is there someone helping with this problem?

Regards

atc1441, 2021-10-05:
Comparing the old and new pcb  we can see they use a new lcd controller ic. 

So that means we need to adapt the display communication

vanMacG, 2021-10-05:
Would better images of the chip help, so you can read the printing?

And besides the display, the measurement seems also broken...

atc1441, 2021-10-05:
Pictures would help to, mostly these chinese chips are not known/without datasheet but we can try

vanMacG, 2021-10-05:
[image]
I'm sorry for the image quality, I only have smartphone/tablet camera. I checked with a magnifying glass:
MCBSLM01
V5K000EG3

Reimund-Koenig, 2021-10-05:
Hello everybody,

there is a funny Information:  Package of mine showed same date "2021.08" (see 3rd picture) but when I opened up my case, I realized that I have HW-B1.6 (2021-03) and not HW-B1.9 (2021-05). 
They change their HW very fast. Chip on the first picture has no description at all :-(
TeLink chip on the 2nd picture is at least the same.
3rd Picture shows the package.

Question: Should I re-open https://github.com/pvvx/ATC_MiThermometer/issues/130 because it is a different HW? Or do we manage it in one issue, because the HW behaves the same?

[image]
[image]
[image]

Reimund-Koenig, 2021-10-07:
Good morning, 

I at least found a way to recover the Mi-Thermometer.

- Connect though it with [pvvx TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)
- Click Original-FW
- Click Flash to Device
- Go to the Xiaomi Home App
- Connect your device (still possible)
- Go to Settings and select "Firmware Update"
- Done --> You can use your device again :)

Cheers Reimund

atc1441, 2021-10-07:
So that means they made a detection for what display controller they got in a device with just one firmware.

This is good as we can see how they detect it and copy it, as in the first model there was no way to read from the display controller just writing.

Can you maybe dump the firmware update file link? You need to be rooted if i remember correctly

atc1441, 2021-10-07:
Took a while to get it but here we go
New firmware attached as reference
[LYWSD03MMC_0130_upd_miaomiaoce.sensor_ht.t2.zip](https://github.com/pvvx/ATC_MiThermometer/files/7300157/LYWSD03MMC_0130_upd_miaomiaoce.sensor_ht.t2.zip)

atc1441, 2021-10-07:
Update and downgrade from an to 0130 works fine. So can be developed with now and reversed further

vanMacG, 2021-10-08:
Sounds great. Unbricking the device with webupdater and the zip you provided here worked.
If you need testers, let us know ;)

H4R0, 2021-10-10:
I got mixed batches where you order for example 4 sensors at once.

Half of them got bricked and I was able to unbrick them with the given stock firmware.

Note that for me not only display was not working but also temperature reports.

BLE communication was the only thing that still worked.

pvvx, 2021-10-10:
> So that means they made a detection for what display controller they got in a device with just one firmware.

Display controller on I2C:
[image]

DeBuXer, 2021-10-11:
> Note that for me not only display was not working but also temperature reports.
> 
> BLE communication was the only thing that still worked.

Same issue here, only in my case it is also not possible to receive the settings from the device like explained [here](https://github.com/pvvx/ATC_MiThermometer#configuration).

I got 6 of these new devices. So if I need to test something, let me know! :)

vanMacG, 2021-10-11:
> PS: I don't have new versions of devices yet :(

Do you need the vendor where I bought mine? V1.9. (German ebay, but shipping took 1 week...)

mariuszkorba, 2021-10-12:
Nice one; how can I chip in as well?
BTW. Great work, you and @pvvx 💪

I got mine devices from here: https://www.aliexpress.com/item/4000427410383.html, and they arrived, albeit to the UK, reasonably quickly.

Good luck with getting yours soon.

MrTheoW, 2021-10-12:
@atc1441 I got two spares laying here. Other two in this 4-pack are already in use without any hiccups.
Will check the versions and contact you if it are the newer versions. Shouldn't take too long from the Netherlands to get it to you.
Btw... You guys are awesome! Thanks for sharing this with us and for all the effort you guys are putting into it.
🙏

atc1441, 2021-10-13:
@MrTheoW 
Thank you. Lets wait a bit to see what version i get  got the one on order and one being send to me so maybe this is already enough

jovannig, 2021-10-15:
> Good morning,
> 
> I at least found a way to recover the Mi-Thermometer (screen and sensors).
> 
> * Connect though it with [pvvx TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)
> * Click Original-FW
> * Click Flash to Device
> * Go to the Xiaomi Home App
> * Connect your device (still possible)
> * Go to Settings and select "Firmware Update"
> * Done --> You can use your device again :)
> 
> Cheers Reimund

I tried but on my page I don't see any "Original Firmware" to click. The only thing I can do is to download the original firmware and flash it. But I don't know where to find the firmware file.

pvvx, 2021-10-16:
I have made several orders in different stores and I am awaiting receipt. Hopefully new versions will come ...
When I receive new versions in a few days, I will build support. There is nothing complicated here.
The main thing is that the new version HW is at hand...

bjeanes, 2021-10-17:
I ordered from https://www.aliexpress.com/item/1005001626975979.html and the 7 (of 10) which I have set up were all v1.9 hardware. The other 3 may also be, but I haven't opened the box.

S0krates23, 2021-10-17:
> Maybe someone can send me one as well? As there are two new versions and not known what some will get

Sure, in Berlin? Tell me where, got 4 here.

atc1441, 2021-10-17:
> > Maybe someone can send me one as well? As there are two new versions and not known what some will get
> 
> Sure, in Berlin? Tell me where, got 4 here.

Near Hamburg, got the B1.6 Version now so only if you may got the B1.9 version it is needed anymore :)

atc1441, 2021-10-17:
You can tell the version via the App nRFConnect

Looking at Hardware revision string:
[image]

S0krates23, 2021-10-17:
Thx for Information, got only 1.6

pvvx, 2021-10-17:
Today I received from the store version B1.4 again (date on the package is 2020-10). We'll have to wait for the next order ... Lottery :(

[image]

pvvx, 2021-10-17:
Original-FW in TelinkMiFlasher updated to 0130.

* Connect though it with [pvvx TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)
* Click "Show all mi keys" -> "!Erase all Mi Keys!"
* Click Original-FW

TimeTravel-0, 2021-10-18:
Sensirion technical customer support: SHV4 datasheet is under NDA, support only for customers (this confirms SHV4 is a Sensirion device).

> unfortunaly i am unable to open the screws. very small torx needed -.- can somebody tell which one exactly?

Screws are Torx T4 or Torx T5.

BasicCPPDev, 2021-10-19:
@atc1441
Look like its not a general rule i am not able to do it with an 1.9 board
  [image]

TimeTravel-0, 2021-10-19:
@BasicCPPDev what happens if you tap on the arrow(s)? :-)

atc1441, 2021-10-20:
Sure it doed work with mi home with it being 0000
 But what i mean does not work is to know via BLE what hardware version you got with it being 0000

florian1984, 2021-10-21:
did you already ordered the new devices from ali? received today 4 of them could donate you one. Im from Zurich but going tomorrow to germany, could you send you one if you want.

[image]

Just contact me if you're intrested.

bbosilje, 2021-11-03:
My issue with applying the alternate firmware to the versions I have is resolved. I am going to close the issue.

Thank you all!

closed 2021-11-03 by bbosilje

pvvx, 2021-11-06:
I got it with aliexpress B1.9. Date on the package: 2021.09.
 One of these days I will do software optimization and consumption tests.

pvvx, 2021-11-06:
> @pvvx Average current with my custom firmware on: B1.4 is 15uA B1.6 is 20uA
> 
> Average current with your custom firmware on: B1.4 is 14uA B1.6 is 16uA
> 
> Average only deep sleep area is: B1.4 is 5.62uA B1.6 is 7.93uA
 
Average current with **origial** firmware v130 on B1.9 - 18..20 uA.
Extended ADV test periods, without measurement transfer unit - 19.2 μA
[OriginalPower](https://pvvx.github.io/ATC_MiThermometer/OriginalPower.html) -> Main loop in Advertising mode (B1.9)

pvvx, 2021-11-06:
@TimeTravel-0 
> Regarding sensor readout delay: What about removal of the delay and just read out prior to every "run measurement" command (adds readout delay of the measurement interval, but temp&hum rarely change fast enough).

For SHV4 sensor in next version sw, if the parameters are set: The sensor is in "low power mode", then the last measurement is read in the next advertising cycle. Those. without waiting for the readiness of the measurement...
[image]
Reduces average consumption not significantly: 14.7 -> 14.5 μA

jddonovan, 2021-11-17:
I suppose this is the best place to add a little info about the SHTV3 vs SHTV4 sensors. Few thoughts on the image (green = SHTV3, yellow = SHTV4):

- Raw values (as attached) suggest that the new sensor is slower to react, but only marginally.
- The clearly visible differences in values seem to be from some sort of internal averaging that uses averaging and hysterisis heavily (compared to previous version). Doesn't go up quite as fast, but reacts instantly when actual temperature turns around
- Much less noise, probably due to internal averaging/hysterisis.

I use my own code for smoothing temperature that employs hysterisis, not averaging. Ie. if temperature is changing in the same direction, apply value. If delta-T (change in temperature) has opposite sign, ignore the value unless difference is bigger than T (assumption that the latest value is just noise). When applying my code on the raw output of the sensors, differences between V3 and V4 dissappear almost completely (better than just mean average). V4 obviously still lagging, but only during large temperature differences. That suggest that the V4 has additional hysterisis post-processing compared to V3 and that the physical sensor membrane might have a larger thermal mass, or it is otherwise slower to respond than V3.

[image]

pvvx, 2021-11-17:
@Jddonovan The PCB version B1.4 does not have a metallization wrap around the sensor...

jddonovan, 2021-11-18:
@pvvx thank you for the clarification! And I entirely missed the metallization.

I'v soldered 40µF caps on around 10 of the devices of the 60+ I've had. Caps don't seem to have much impact on the noise, but it might also be due to TX being set at -3db -> difference will no doubt show if power is increased. Caps are there for tx +1db and to squeeze the last juice out of the CR batteries. 

<details>
  <summary>Here are two other sensors, this time from the bathroom, yellow SHTV4, and some additional thoughts **(click to expand)**</summary>
  
[image]

For the life of me I can't explain the lagging with anything else than sensor's internal code changes, ie. extra processing. Devices are side by side with same firmware & settings. Just a heads up for those who are using this for automation. 

But by far the biggest factor for the "noise" is actual real temperature micro-fluctuations that will be present practically in any room caused by air exchange systems or pressure differences from the wind from outside. Figuring how to get the temperature constant is incredibly hard – I put few sensors to be calibrated in a box with thermal mass inside for baseline noise.

All this tinkering is months ago, and again, more practical than scientific. In any case, we're typically talking tiny differences (0.01C). My biggest reason to optimise (smooth) the output is to save database space but still being able to record as accurately as possible.

[image]
Works great with AA/AAA batteries too!

[image]

</details>

I use these sensors, amongts other things, in an industrial metal workshop in the cold Finnish climate. When one of the 4 loading doors are opened, the temperature drop gets detected rapidly by the sensor devices and sent to MQTT. There are also multiple sensors installed close to the work stations. I have a ESP32 chip controlling the HVAC system to control air flow and temperature, effectively improving energy efficiency and comfort. It also reacts to how many people are logged in as working at the site (max 12 ppl) and to dust and VOCs. It's really more of a hobby, but with real tangible results.

Thank you so much @atc1441 and @pvvx! Without you guys we wouldn't have these cheap devices and dare to experiment.

shaaati, 2023-01-08:
I noticed a weird behaviour that threw me off for quite some time:
It seems that weak/bad batteries can result in a similar behaviour to what is described in this thread.
Two devices running on firmware 3.8 seemed to be frozen (display didn't update with new values and devices were missing in Home Assistant, display can easily be checked by covering the measurement hole with a finger for a few seconds).
Then, when I removed and re-inserted the battery, the display stayed blank and the devices seemed to be "dead" (no Bluetooth, nothing in HA).

Turns out when you insert a good and healthy battery (it seems I bought a bunch of extraordinarily bad ones) they work again as if nothing happened.
Just reporting here in case anyone else has similar problems and stumbles upon this issue just as I did.

hpnnet, 2025-07-07:
Help to revive LYWSD03MMC B1.9 via USB COM. I have already tried different firmware, methods of connecting wires, upload speed. If you have experience, please write. 

P.S. I have already revived version 1.5, it did not work on version 1.9.