pvvx/ATC_MiThermometer #126: Thread/Matter support?

KenMacD, 2021-09-28:
Just curious, any plans to add Thread and/or Matter protocol support?

pvvx, 2021-09-29:
There is no support, technical information, or devices from the Connectivity Standards Alliance for users.

KenMacD, 2021-09-29:
There seems to be some demo code for the TLSR9-series chips at https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/telink, and some 802.15.4 code already at http://wiki.telink-semi.cn/wiki/chip-series/TLSR825x-Series/.

Part of why I'm asking is that I already have an OpenThread network deployed and listening for BLE broadcasts doesn't seem like it'll play nicely with it.

KenMacD, 2021-09-30:
To check if Telink might be planning support I asked [over at their forum](https://developers.telink-semi.cn/topic/1034):

Me:

> Hello, I see the code for the 9x versions, but I was wondering if there's any plans to support the older series chips with this new protocol.

Telink:

> Support for matter is only available on TLSR9.

Where the chip seems like it _should_ support 802.15.4 I'm guessing that's not a hard limitation, just that they don't have anything available for it.