pvvx/ATC_MiThermometer #129: CGG1 (old one) won't connect

Notrial, 2021-10-04:
Is there a way to connect to the old CGG (the one without the bindkey)?
When I try to connect it just times out with: _Something went wrong, to many reconnect's_

przemhb, 2021-11-10:
Have you succeeded in connecting?
I've got the same issue:
```
22:35:46: Connecting to: ClearGrass Temp & RH
22:35:55: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
22:35:55: Reconnect 1 from 5
22:35:55: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
22:35:55: Reconnect 2 from 5
22:35:55: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
22:35:55: Reconnect 3 from 5
22:35:55: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
22:35:55: Reconnect 4 from 5
22:35:55: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
22:35:55: Reconnect 5 from 5
22:35:55: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
22:35:55: Something went wrong, to many reconnect's
```

How to fix the issue or at least workaround it? Anyone?

Notrial, 2021-11-11:
Nope, doesn't work unfortunately :)

closed 2021-12-12 by pvvx