pvvx/ATC_MiThermometer #130: Blank Screen

Reimund-Koenig, 2021-10-05:
Hi @pvvx, first of all: Thank you for your work.

My Display is blank since I flashed "/atc1441/ATC_MiThermometer" latest release to my brand new LYWSD03MMC (2019DP8115) device. Connecting, flashing OTA is working. Advertising and display not.

Flashed back "original FW" and your latest custom FW (ATC_Thermometer32z.bin). Display stayed blank. 

Also tried "Repair LCD" button with your latest FW which returns "7:06:02 AM: Send 'Repair LCD' ok" but display stayed blank.

Still connecting, flashing OTA is working. Advertising and display are broken.
Is there a way to recover the display and advertising?
(Is there more hope for recovery if I flash via com-adapter?)

Cheers
Reimund

Reimund-Koenig, 2021-10-05:
Here some logs:

7:17:45 AM: Load firmware file 'ATC_Thermometer32z.bin'...
7:17:45 AM: File size: 62708 bytes
7:17:45 AM: Count: 3920
7:27:22 AM: Settings 23 was send successful
7:27:23 AM: Device Date: 1970-01-01 00:21:57.000
7:27:27 AM: Send new DevTime ok
7:27:28 AM: Settings 23 was send successful
7:27:28 AM: Device Date: 2021-10-05 07:27:27.000
7:27:28 AM: Device Date: 2021-10-05 07:27:27.000
7:27:51 AM: Settings 22 was send successful
7:27:53 AM: ExtShow: big_number: 0, small_number: 0, ext.vtime: 600 s, flg: AF
7:27:56 AM: Send 'Repair LCD' ok
7:28:08 AM: Settings 33ff was send successful
7:28:12 AM: Settings 3300 was send successful
7:28:24 AM: Settings 55 was send successful
7:28:26 AM: Hardware Version: LYWSD03MMC, Software Version: 3.2
7:28:26 AM: Custom config: [5, 0, 0, 0, 40, 4, 169, 124, 49, 56, 60]
7:28:57 AM: Send Ext data ok
7:28:58 AM: ExtShow: big_number: 1234, small_number: 99, ext.vtime: 600 s, flg: C7
7:28:59 AM: Settings 22 was send successful
7:29:01 AM: ExtShow: big_number: 1234, small_number: 99, ext.vtime: 600 s, flg: C7
7:29:10 AM: Send cmd GetMemo ok
7:29:11 AM: Memo end: 0
7:30:58 AM: Send 'Send clock data on LCD' ok
7:31:05 AM: Send 'Repair LCD' ok
7:31:05 AM: Send 'Repair LCD' ok

Still have a blank display without any reaction :(

Reimund-Koenig, 2021-10-05:
I was brave and flashed a second device OTA.
Same issue > Display is blank :-(
Seems like they did some changes to prevent flashing custom Firmwares under the same CMITID (2019DP8115) and name (LYWSD03MMC).

vanMacG, 2021-10-05:
See #125

atc1441, 2021-10-05:
Are you able to dump the firmware of a stock one?

vanMacG, 2021-10-05:
> Are you able to dump the firmware of a stock one?

Is this possible OTA? I still have some new ones with stock firmware.

Maybe @Reimund-Koenig can help if adapter is needed:
> (Is there more hope for recovery if I flash via com-adapter?)

Reimund-Koenig, 2021-10-05:
Hello, 

@atc1441 I am not able to dump the firmware without help (small tutorial)
@vanMacG I have one more adapter left (on my desk). But I can order new ones (if needed)

--> Lets continue our conversation here  https://github.com/pvvx/ATC_MiThermometer/issues/125 :)

closed 2021-10-05 by Reimund-Koenig