pvvx/ATC_MiThermometer #144: Wrong firmware - Bricked or not bricked?

bolovanos, 2021-11-11:
Hi, I have just used [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) on my  LYWSD03MMC (ver 1.4), flashing it with [MHO_C401_v34c.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/MHO_C401_v34c.bin). 
Devices display is dead - shows nothing, [advertising on my ESP32](https://github.com/pvvx/ATC_MiThermometer/tree/master/esp32/Advertising) and TelinkMiFlasher (mobile chrome ) no longer see it.

Is there a way to flash it back or is it in silicone heaven now?

bolovanos, 2021-11-12:
Will answer myself - [How to flash the custom firmware or unbrick the device](https://github.com/atc1441/ATC_MiThermometer#how-to-flash-the-custom-firmware-or-unbrick-the-device). 

What if third person decides to brick these sensors with Telink flasher? Is there a protection from this "attack"?

mdarii, 2021-11-12:
@bolovanos to avoid that you can use pincode option

bolovanos, 2021-11-12:
@mdarii Hi. You mean with new firmware or is it possible with original one?

mdarii, 2021-11-12:
@bolovanos not sure what you mean by new, but I think Xiaomi original firmaware also allow using of pincode, custom firmware also supports it

atc1441, 2021-11-12:
@bolovanos  there is no need to write me a private email about this problem :)

You can ask it here

bolovanos, 2021-11-12:
@mdarii New - custom firmware.  Not sure if I saw that option in [TelinkFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).
Btw - does it mean that these sensors are prone to this "attack" when used directly?

@atc1441 Right - sry :)
1) What torx screwdriver is used on LYWSD03MMC? The one I have T06 is a bit wider.
2) [Wiring you use  tx/rx](https://github.com/atc1441/ATC_MiThermometer/blob/master/Mi_SWS_Connection.jpg) -  is wire connection from P10 the way it is on this scheme(taken from pvvx page) ?
[image]

pvvx, 2021-11-22:
@bolovanos 
For a PC, there is a firmware option via a USB-COM adapter in the Chrome explorer.
https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version
Only the TX adapter output and SWS pin SoC TLSR825x are used.

The pin-code only works on custom firmware.

yeyeto2788, 2021-11-24:
Hey guys!

I did tried to flash the firmware using the python script but the device is still bricked 😢 

Here you can see that it is supposed to be working and flashed with [newest](https://github.com/pvvx/ATC_MiThermometer/blob/master/ATC_V34d.bin) but in reality nothing is happening which is really sad for me.
```console
yeyeto2788@juan-thinkbook:~/Downloads$ python3 ATCtelink.py ./ATC_V34d.bin 
=======================================================
TLSR825x Flasher, modified from ComSwireReader Utility version 00.00.01
-------------------------------------------------------
Open /dev/ttyUSB0, 921600 baud...
UART-SWS 92160 baud. SW-CLK ~23.5 MHz(?)
Erasing flash ...
Started writing 0xf8a4 bytes
Writing at 0x0 (0%)
Writing at 0x100 (0%)
Writing at 0x200 (0%)
Writing at 0x300 (1%)
Writing at 0x400 (1%)
Writing at 0x500 (2%)
Writing at 0x600 (2%)
Writing at 0x700 (2%)
Writing at 0x800 (3%)
Writing at 0x900 (3%)
Writing at 0xa00 (4%)
Writing at 0xb00 (4%)
Writing at 0xc00 (4%)
Writing at 0xd00 (5%)
Writing at 0xe00 (5%)
Writing at 0xf00 (6%)
Writing at 0x1000 (6%)
Writing at 0x1100 (6%)
Writing at 0x1200 (7%)
Writing at 0x1300 (7%)
Writing at 0x1400 (8%)
Writing at 0x1500 (8%)
Writing at 0x1600 (8%)
Writing at 0x1700 (9%)
Writing at 0x1800 (9%)
Writing at 0x1900 (10%)
Writing at 0x1a00 (10%)
Writing at 0x1b00 (10%)
Writing at 0x1c00 (11%)
Writing at 0x1d00 (11%)
Writing at 0x1e00 (12%)
Writing at 0x1f00 (12%)
Writing at 0x2000 (12%)
Writing at 0x2100 (13%)
Writing at 0x2200 (13%)
Writing at 0x2300 (14%)
Writing at 0x2400 (14%)
Writing at 0x2500 (14%)
Writing at 0x2600 (15%)
Writing at 0x2700 (15%)
Writing at 0x2800 (16%)
Writing at 0x2900 (16%)
Writing at 0x2a00 (16%)
Writing at 0x2b00 (17%)
Writing at 0x2c00 (17%)
Writing at 0x2d00 (18%)
Writing at 0x2e00 (18%)
Writing at 0x2f00 (18%)
Writing at 0x3000 (19%)
Writing at 0x3100 (19%)
Writing at 0x3200 (20%)
Writing at 0x3300 (20%)
Writing at 0x3400 (20%)
Writing at 0x3500 (21%)
Writing at 0x3600 (21%)
Writing at 0x3700 (22%)
[... 174 lines cut ...]
Writing at 0xe600 (92%)
Writing at 0xe700 (92%)
Writing at 0xe800 (93%)
Writing at 0xe900 (93%)
Writing at 0xea00 (94%)
Writing at 0xeb00 (94%)
Writing at 0xec00 (94%)
Writing at 0xed00 (95%)
Writing at 0xee00 (95%)
Writing at 0xef00 (96%)
Writing at 0xf000 (96%)
Writing at 0xf100 (96%)
Writing at 0xf200 (97%)
Writing at 0xf300 (97%)
Writing at 0xf400 (98%)
Writing at 0xf500 (98%)
Writing at 0xf600 (98%)
Writing at 0xf700 (99%)
Writing at 0xf800 (99%)
Done, resetting CPU
```

Even plugging and unplugging the device does not work. Any way to fix that? 

BTW, I flashed by mistake the [MHO_C401_v34c.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/MHO_C401_v34c.bin) due to the fact that yesterday there was a new change introduced on the HTML and it couldn't retrieve the `V34c` for the `LYWSD03MMC`

atc1441, 2021-11-24:
Please try flashing it a few times and or try a different usb uart converter

yeyeto2788, 2021-11-24:
Got it! Thanks for the reply @atc1441 Should I also try flashing the stock firmware?

atc1441, 2021-11-24:
Yes flashing stock should also work

yeyeto2788, 2021-11-24:
@bolovanos this are the connections I used
[image]

I hope it helps you.

atc1441, 2021-11-24:
Reset is wrong if i remember right. Thats just a gpio.
But i could be wrong its put of my head.

yeyeto2788, 2021-11-24:
@atc1441 then that would mean that I would need to unplug and replug the USB-Serial converter in order to hard reset the device or is it needed for something else?

atc1441, 2021-11-24:
Lift up the battery + pin shortly  :)

yeyeto2788, 2021-11-24:
[image]

Finally working! Wujuuuuu

Thanks for the help!

atc1441, 2021-11-24:
Nice :)

Bitman-GR, 2021-11-24:
What probes are you using with these needle pins ?

> @bolovanos this are the connections I used [image]
> 
> I hope it helps you.

yeyeto2788, 2021-11-24:
> What probes are you using with these needle pins ?
> 
> > @bolovanos this are the connections I used [image]
> > 
> > I hope it helps you.
> 
> 

Those are regular sewing needles (cut with pliers) with wrapping wires I'm too cheap to buy specialized tools. The rig is 3D printed BTW

mdarii, 2021-11-24:
> > What probes are you using with these needle pins ?
> > > @bolovanos this are the connections I used [image]
> > > I hope it helps you.
> 
> Those are regular sewing needles (cut with pliers) with wrapping wires I'm too cheap to buy specialized tools. The rig is 3D printed BTW

@yeyeto2788 could you share sources for the rig? looks very useful when required to connect without soldering

yeyeto2788, 2021-11-24:
@mdarii [here](https://www.thingiverse.com/thing:2111631) you have. They work like a charm and I haven't even print all of the pieces.

mdarii, 2021-11-24:
> @mdarii [here](https://www.thingiverse.com/thing:2111631) you have. They work like a charm and I haven't even print all of the pieces.

@yeyeto2788 thank you

evildad, 2021-11-26:
@yeyeto2788 Which USB to UART Converter did you use?

yeyeto2788, 2021-11-26:
@evildad > FTDI USB to serial converter

[image]

I bet most 3.3v USB to serial converter would work. Even a CH340

pvvx, 2021-11-26:
FTDI is not suitable for Telink S-Wire emulation... 
CH430, CH340c, Prolific PL-2303HX, ...

bolovanos, 2021-12-02:
Finally I have needed components / tools.

@pvvx
"For a PC, there is a firmware option via a USB-COM adapter in the Chrome explorer. ..."
- I am a bit lost in schematics. For browser flashing - Can I use version B of my schematics, or C mentioned by you in your link ([The USB-COM adapter writes the firmware in explorer. Web version.](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version) "Connect only TX-SWS and GND wires."
[image]
- Version A is suitable for [TlsrComSwireWriter](https://github.com/pvvx/TlsrComSwireWriter) or for ATCtelink.py by @atc1441 - right?
- That Reset is not needed in my case?

@yeyeto2788
- Thank you - that RST is a bit confusing - place where to connect on ATC is somehow "clear" see ^ - but from your picture is not clear whether you use that pull up resistor and how you work with that reset (which seems to be not needed). Btw nice rig - I thought to ask you for source, but I noticed texture of 3D printer which I do not have. :)

@atc1441
"Lift up the battery + pin shortly :)" You mean that one and only p14 pin? :)

pvvx, 2021-12-03:
A, B and C are fine.
In some cases, either a reset signal or a power on is required. These signals are taken from the adapter's DTR or RTS pins.
The controller reset signal is not output to the test point. It is possible to use the output of the capacitor connected to leg 18 "RESETB" TLSR8251F512ET24. It is possible to short-circuit "RESETB" to GND manually, while "ATime" is working - activate the time

In the case of power switching, this is done:
1. Manually, while "ATime" is running - activate time
2. Through the key from the DTR or RTS adapter.

Такие случаи возникают, если контроллер завис или находится в спящем режиме, включена защита Telink от прошивки :)

closed 2023-02-18 by pvvx