pvvx/ATC_MiThermometer #145: LYWSD03MMC new version B1.7

ckyOL, 2021-11-17:
The production date of my xiaomi thermometer is 202110. When i try to flash it, i find that it is B1.7. Does the latest firmware supports it? 

the packaging picture
[image]

PCB pictures
[image]
[image]

jddonovan, 2021-11-17:
Not to my knowledge. These appear to be rare. I'm guessing the sensor is new (shtv4) and same as 1.6 &1.8 but the display controller is same as v1.4. As such it appears to be a mix and everything is supported (I assume). There is an autodetect for the version. I don't think there's any autodetect for individual components, so it probably requires a few lines of code. 

It is likely that you can try flashing it. It's unlikely it will brick completely if the main chip and (possible) flash chip remain the same.

atc1441, 2021-11-17:
Both PVVX and my version use autodetect per component, so this version should be fully supported.

jddonovan, 2021-11-17:
> Both PVVX and my version use autodetect per component, so this version should be fully supported.

Good to know!

@ckyOL could you confirm if it works ok? If it does, it seems relatively safe to add this device to the list of supported versions.

ckyOL, 2021-11-17:
@jddonovan It seems work fine after i flashed it and changing some config.
[image]
 
Interestingly, the hardware revision string was changed to B1.4
[image]
.

jddonovan, 2021-11-17:
@ckyOL I've looked at the code how it detects the revision, but frankly I cannot remember anymore. It's indeed captured before flashing as presented in the logs. But after flashing, it might autodetect it by e.g. the chip used for the screen..? Also possible that the stock version number gets stored at the time of flashing, which means that the devices flashed are "stuck" showing the wrong version now.

Regardless, if everything works and the temperature sensor autodetection code remains autodetected and not based on firmware, I think you're ok. However, low priority patch for the version number is still warranted.

ckyOL, 2021-11-17:
It is able to flash origin firmware safely.  And hardware version string 

> 
> 
> @ckyOL I've looked at the code how it detects the revision, but frankly I cannot remember anymore. It's indeed captured before flashing as presented in the logs. But after flashing, it might autodetect it by e.g. the chip used for the screen..? Also possible that the stock version number gets stored at the time of flashing, which means that the devices flashed are "stuck" showing the wrong version now.
> 
> Regardless, if everything works and the temperature sensor autodetection code remains autodetected in the future as well, and not based on firmware, I think you're ok.
> 
> However, low priority patch for the version number is still warranted.

I flashed origin firmware and hardware version string came back to B1.7 correctly.

pvvx, 2021-11-18:
@jddonovan > Could you confirm the chip remains the same? (TLSR8251F512ET24)

Communication and the ability to reprogram remain on all versions of TLSR8251Fxx, TLSR8253Fxx, TLSR8258Fxx.
These chips have the same crystal. The difference is in contacts and packaging.

ckyOL, 2021-11-18:
Yes, it is TLSR8251F512ET24.
[image]

pvvx, 2021-11-21:
@ckyOL 
I do not have HW: B1.7 and B1.6 in stock. To search and test various options, I already ordered LYWSD03MMC from many suppliers and received dozens of HWs: B1.4 and B1.9 with different release dates. I have not found any other versions for today. The stores do not provide types of hardware version ...
You can tell that TelinkMiFlasher.html writes in the log line: "Hardware version: LYWSD03MMC B1.?, Software version: 3.4, Sensor :**?**"

ckyOL, 2021-11-22:
> @ckyOL I do not have HW: B1.7 and B1.6 in stock. To search and test various options, I already ordered LYWSD03MMC from many suppliers and received dozens of HWs: B1.4 and B1.9 with different release [...]

It shows that " Hardware Version: LYWSD03MMC B1.4, Software Version: 3.4, Sensor: SHT4x".  The hardware revision string was changed to B1.4 after flashing v3.3 firmware. And it was still B1.4 after updating to v3.4.

pvvx, 2021-11-22:
Thanks. This makes it clear that version B1.7 uses an SHT4x type sensor and the display is controlled by the IST3055NA0 microcircuit.
In the next version I will fix the HW variant...

closed 2021-12-03 by ckyOL