pvvx/ATC_MiThermometer #146: Config option do not appear

delijati, 2021-11-21:
Hi,

i flashed the v34 and v33 version in both i can't get the config to appear. As far as i understand the Flasher JS code it is the `devidEnabled` variable it has to be `0000181a-0000-1000-8000-00805f9b34fb` but i have `00010203-0405-0607-0809-0a0b0c0d1912`.

Do i only have to add this id `00010203-0405-0607-0809-0a0b0c0d1912` to the JS Code?

Thanks

More Info:
[image]

pvvx, 2021-11-21:
I can't understand the question. 
According to the logs presented, you have a bad connection... The device is not responding. There can be many reasons.

delijati, 2021-11-21:
Ok found the error :/ I have 4 of this sensors and when they are to close to each other the connections seams to to drop. Sorry for the noice and thanks for the hint :)

closed 2021-11-21 by delijati