pvvx/ATC_MiThermometer #147: Questions: ESP32 with original mi firmware

kiralikbeyin, 2021-11-22:
I searched a lot but couldn't find.

Is there any example esp32 sketch (no esphome or hass etc) which directly gets mi data (no custom firmware, no activation etc..)?

Esp32 example of this lib only show;
22:40:12.709 -> Found service 'fe95' data len: 16, 0f1695fe30585b051439375838c1a408
22:40:12.709 -> MAC: a4c138183739
22:40:12.709 -> count: 20

pvvx, 2021-11-22:
I have not seen this in open sources.
ESP32 has flaws in BLE libraries for ad scanning. There are a lot of missing packets...

kiralikbeyin, 2021-11-23:
> I have not seen this in open sources. ESP32 has flaws in BLE libraries for ad scanning. There are a lot of missing packets... PS: To enable [relays from external BLE sensors](https://github.com/pvvx/AdScanerTrg), I use cheaper BLE modules.

Nice work.

I am not good with hardwares, simply esp32 with bluetooth is more easy to use.

Is there any document about how to decode original firmware code (or is it possible) ?

22:40:12.709 -> Found service 'fe95' data len: 16, 0f1695fe30585b051439375838c1a408

pvvx, 2021-11-23:
https://github.com/pvvx/ATC_MiThermometer/blob/master/esp32/Advertising/Advertising.ino

kiralikbeyin, 2021-11-24:
> 

I tried the link thank you.

Both codes need "custom firmware" for mi.

I am searching a decoder code (human readable temp-hum) for original mi firmware.. (advertising ok or subscription ok)

Is that possible or a fantasy?

pvvx, 2021-11-24:
There is nothing complicated there. You need to add AES CCM decryption.
But everyone has different AES libraries...
Python example:
https://github.com/pvvx/ATC_MiThermometer/blob/master/test_adv_key.py

closed 2022-05-01 by pvvx