pvvx/ATC_MiThermometer #148: Question: Remove screen with 123.4?

stratus-ss, 2021-11-22:
Hi,

First thanks for all your efforts. Its really great to see this project active.

I have flashed my first device successfully but I don't understand the configuration options. I set the temp, removed the smiley, added the battery. All that works great, but there is a screen that rotates through that has "123.4" sample data. 

How do you remove this? Is the firmware hard coded to loop through 3 screens and if it doesn't have a screen it shows this data?

Can you explain the purpose and how to get rid of this screen?

Thanks!

pvvx, 2021-11-22:
The screen with custom values is displayed for a limited time set in the settings:
[image]

stratus-ss, 2021-11-22:
Thank you for the reply. I didn't get the tool tip popup. So the 600 is the number of seconds it is displayed?

closed 2021-11-24 by stratus-ss

iondarie, 2022-01-05:
@stratus-ss did you manage to turn that screen off?

stratus-ss, 2022-01-05:
As the author mentioned this screen eventually goes away on its own. Its meant for testing purposes, although I personally haven't found a use for it