pvvx/ATC_MiThermometer #149: Building a Bridge

CSchoch, 2021-11-23:
Hi,

Which Hardware as Esp32 replacement would you use to build a proper Bridge to Wlan or Lan.
Considered the high loss of esp32. 
Kr
Christian

pvvx, 2021-11-23:
ESP resources are insufficient to serve the full TCP / IP networks specification. For toys only.
For normal operation on IP networks, they use Unix-like operating systems based on the Linux kernel ...

ikke-t, 2021-12-18:
For the reference, I use Raspberry Pi with bluewalker sniffing the broadcast data, and pushing it to telegraf socket, which sends them to influxdb. It's often overkill, but I have the Raspis anyhow in my home automation setups.

nagyrobi, 2022-03-19:
ESP32 is known for high loss because it uses the same radio module for BLE and WiFi in a shared way.

You can get very good BLE performance on ESP32 if you're not using WiFi at all, but use an Ethernet module instead. Look for Olimex ESP-32-POE, WirelessTag WT32-ETH01, TTGO T-Internet-POE ESP32 or similars. 

Run ESPHome with Ethernet configuration and use `esp32_ble_tracker` with `pvvx_mithermometer` sensors from the doc example.

pvvx, 2022-03-19:
Tests have shown the ESP32 to be the worst of the few SoCs when running BLE only. When testing, the transfer of the counter of received advertising packets to the UART was used.

pvvx, 2022-03-19:
ESP32 fails to create a "cascaded receiver-repeater" with battery power.
On cheaper TLSR825x chips, the consumption when accompanied by receiving and transmitting from Xiaomi LYWSD03MMC was 70..75 μA.
[image]

An unfolding of some cycles of automatically synchronized transmission-reception of advertisements:
[image]
Max Cycle:
[image]
Advertising in BLE is transmitted with random delay insertion to reduce collisions. From this, the receive window has a different duration.

When testing such a cascaded system with automatic transmission-reception synchronization, not a single packet was lost per hour. 
On newer BLE SoCs, the consumption is even lower and it is possible to use the CR2032 to work for more than 8 months.
And what is the purpose of using an ESP32 that requires a 1A power supply?

nagyrobi, 2022-03-19:
I really don't care about how much power eats an ESP32... it's still way less than a raspberry for example, which also can have a lot more other problems on long term like failing SD cards and boot issues after long operation times. Having a full blown linux running up just to gather data from BLE devices and forward them on to another system is an overkill and source for operational troubles on long term.

ESP32 agree it's not perfect, but it's a kind of set it and forget it solution which _can be_ reliable enough for most people, with a much longer lifespan than a full-blown operating system solution will ever have.

nagyrobi, 2022-03-19:
> ESP32 is the last century of electronics...

I consider this as an advantage because many people know it well, has great open source community support and knowhow is very wide.

Agree that newer chips are more modern and can do much more in a much better way but are still kind of exotic for the majority of the users. It will take more time to get widely supported by open source projects like ESPHome or Tasmota, and by that time they will be also considered antique. It's an endless loop :-)

pvvx, 2022-03-19:
> Меня действительно не волнует, сколько энергии потребляет ESP32 ... это все же намного меньше, чем, например, малина, у которой также может быть гораздо больше других проблем в долгосрочной перспективе, таких как сбои SD-карт и проблемы с загрузкой после долгого времени работы. Наличие полноценного Linux, работающего только для сбора данных с устройств BLE и пересылки их в другую систему, является излишним и источником операционных проблем в долгосрочной перспективе.

Даже древний RT5350 с активным Ethernet потребляет одинаково с ESP32.
[image]

1. CPU без MMU не может работать стабильно с динамическим распределением памяти из-за дефрагментации.
2. TCP стек требует буфера в памяти для самого стека и памяти TIME_WAIT структур. Это не менее 0.5 Мегабайта.

BLE тем прорвало, что не требуется поддержка TCP и прочих сетевых протоколов.

> ESP32 согласен, что это не идеально, но это своего рода решение «установил и забыл», которое _может быть_ достаточно надежным для большинства людей, с гораздо более длительным сроком службы, чем когда-либо будет иметь полномасштабное решение операционной системы.

Такое понятие, как  "Надежность", не рассматривается в Arduino и текущей IoT сфере.
 
> Кроме того, в сочетании с сетевым коммутатором PoE можно управлять всей системой от общего управляемого источника питания, защищенного ИБП и т. д.

Это требует перестройку жилых помещений - дополнительную проводку что и составляет большую стоимость такого решения. И часто просто не применимо в концепции «установил и забыл».

pvvx, 2022-03-19:
> I consider this as an advantage because many people know it well, has great open source community support and knowhow is very wide.
>
> Agree that newer chips are more modern and can do much more in a [...]

It does nothing for the main audience. Using ESP requires tools and specialized knowledge. ESP cannot be flashed from a smartphone.

nagyrobi, 2022-03-19:
Well you probably didn't study deep enough the ESPHome project. Everybody who can deploy a Home Assistant instance by themselves, will be able to auto-generate preconfigured firmwares for any kind of custom ESP solution, one needs just a USB cable (or an USB-TTL adapter at most).

I'm not talking theoretics, I do have my WT32-ETH01 running and forwarding sensor data from LYWSD03MMC to Home Assistant as native sensors.

[image]

Things I've mentioned do matter for this kind of audience, which is not small at all.

pvvx, 2022-03-19:
> Well you probably didn't study deep enough the ESPHome project. Everybody who can deploy a Home Assistant instance by themselves, will be able to auto-generate preconfigured firmwares for any kind of custom ESP solution, one needs just a USB cable (or an USB-TTL adapter at most). Firmware can be flashed directly from the browser.
 

Here I have a ESP32 module from the store. Plugged it into a USB charger. There is a smartphone. How to flash?

nagyrobi, 2022-03-19:
This becomes offtopic ftom the original OP question. I presented a viable solution to @CSchoch and I'm terribly sorry you dont like it - but the fact is, it works and can be implemented relatively simply and cheap. 

My post wasn't intended to offend you in any way, please excuse me if I made it feel that way.

pvvx, 2022-03-19:
I also devoted a lot of time to ESP - more than 5 years. Based on this, I describe the problems that exist there. And the goals are not in the offtopic, but in finding solutions to these issues.

And look around - the main questions and problems occur among ESP users.
People who use other means have no questions - you press a couple of buttons on the phone and everything works.

Protoncek, 2022-03-19:
> >
> 
> Here I have a ESP32 module from the store. Plugged it into a USB charger. There is a smartphone. How to flash?.

Well, if you mean first time - yes, you must connect it via UART and program it. But all later versions go via OTA and thus also via phone - connect to module via IP and upload new version - if that's what you mean. If you mean change only minor things - yes, ESPHome is built in a way that each and every thing requires upload whole bin file again and again... i hate this part, too. Tasmota is more flexible in this regard - it allows many changes via console.
But, yes, ESPHome has problems with BLE. But, why tasmota on ESP32 doesn't? I mean - faaaar less... so, it's not entirely hardware's fault...

nagyrobi, 2022-03-19:
> you press a couple of buttons on the phone and everything works.

I'd be happy if you could post us a link to such a product which can forward BLE data to Home Assistant - nothing more.

pvvx, 2022-03-19:
If you dig deeper, then a typical module is sold with "AT" firmware. Includes OTA. But you need to send the command via USB-UART. And in OTA there is a link to the URL, which can be redirected in the program on the smartphone.
But this does not exclude the connection with a cable to a smartphone that has the ability to work with USB devices.
It is easier for an ordinary programmer to write an analogue of esptool in HTML with javascript.
But here, too, the "ESP community" did not show itself.

nagyrobi, 2022-03-19:
> But here, too, the "ESP community" did not show itself.

@Protoncek I think this shows us this conversation is not objective.

pvvx, 2022-03-19:
As a result, the only remaining solution for BLE is the use of RPi and ready-made analogues like "Home Assistant Blue".

nagyrobi, 2022-03-19:
Well that's not the category of "you press a couple of buttons on the phone and everything works" lol.

Protoncek, 2022-03-19:
> As a result, the only remaining solution for BLE is the use of RPi and ready-made analogues like "Home Assistant Blue".

Hm... i guess that it's true if you intend to do more with BLE. However, just for xiaomi temperature and moisture reading my ESP32 with tasmota reads values quite perfectly. No need for anything else. Also, after reboot (or startup of ESP32) all values are shown in a matter of seconds (unlike when using ESPHome, where getting all data could take as long as 5 minutes.

nagyrobi, 2022-03-19:
@Protoncek under current ESPHome (2022.3 at time of writing) values show instantly after the node boots up.

pvvx, 2022-03-19:
> Well that's not the category of "you press a couple of buttons on the phone and everything works" lol.

Isn't there a pre-installed program in these devices?

nagyrobi, 2022-03-19:
Thanks for the tip, but that costs more than 5 times than an ESP32 Nodemcu, and almost 3 times more than a WT32-ETH01 for this region - but that's again, offtopic.

And there's no _open source_ firmware alternative for it. No chinese cloud managed stuff allowed here, sorry. Has to be able to operate off the cloud from the very first boot.

pvvx, 2022-03-19:
> And there's no open source firmware alternative for it.

https://github.com/AlexxIT/gw3

nagyrobi, 2022-03-19:
Not qualifying: _All Xiaomi devices must be linked to an account via the Mi Home app, otherwise they don't send data._

nagyrobi, 2022-03-19:
@pvvx you can try as hard as you want, you won't convince us.

Protoncek, 2022-03-19:
> @Protoncek under current ESPHome (2022.3 at time of writing) values show instantly after the node boots up.

Is that so? I must test it then. Is there anything that has changed in ESPHome? From my understanding bad drivers are(were?)  the main cause for bad working...
Regarding xiaomi gateway: i don't know it, but my main condition is also - working LOCALLY. Not with cloud.

pvvx, 2022-03-19:
I'm not going to convince. I'm looking for normal solutions, not toys in ESP.

There are many [Tuya Smart Plug](https://www.elektroda.com/rtvforum/viewtopic.php?p=19851328#19851328) devices on the market. They have a BK7231x chip installed. It supports BLE and WiFi. Full replacement ESP32 has the ability to build a BLE-WiFi gateway.
It is possible to flash from a smartphone. For the mass user, the OTA firmware is not yet available, but publication is possible in the future...

Initial Open Source Integration: 
[[BK7231T] My HTTP server, configurator, MQTT support from Home Assistant](https://www.elektroda.com/rtvforum/topic3866123.html)

nagyrobi, 2022-03-19:
ESPHome project starts to support RP2040. I wouldn't be surprised to cover BK7231x too in the future.

pvvx, 2022-03-19:
> ESPHome project starts to support RP2040. I wouldn't be surprised to cover BK7231x too in the future.

The ESPHome project should force sellers to sell ESP32 modules with open OTA firmware. And also supply a power supply and a case. Then it might be worth considering.

nagyrobi, 2022-03-19:
There are products which come with ESPHome preinstalled. Same with Tasmota.

But I wouldn't wanna pay a penny more to have it preinstalled, while I can do it on my own.

pvvx, 2022-03-19:
Coercion is due to popularity, price and consumer qualities. ESP32 does not have this today. Only in narrow specific circles, and not for the mass user.

nagyrobi, 2022-03-19:
Well that's only how you see it.

pvvx, 2022-03-19:
> Well that's only how you see it.

I not only see it, but I can participate and correct this process. This is not the first time and there is a lot of experience in such cases. :)

nagyrobi, 2022-03-20:
[image]

closed 2023-02-18 by pvvx