pvvx/ATC_MiThermometer #151: Does the TelinkOTA tool suitable for any TLSR-based device?

vevsvevs, 2021-11-25:
Good day! Could you please answer some general questions about the Telink FW?

1. Telink OTA is hardware- or software-related stuff? I mean, is it possible to update every TLSR82-based device via the TelinkOTA tool, or OTA should be explicitly supported by the current device FW?
2. Is FW image consistency protected anyhow (checksum/cert), or it can be rudely patched and uploaded without a hitch?

As it turned out, both Yeelight remote and Dimmer are based on the TLSR82 solution. If it would be possible to change their hardcoded PIDs without any soldering - they might be used in the app directly as independent devices, which is really convenient stuff.

Thanks in advance for any info!

pvvx, 2021-11-25:
1. OTA should be explicitly supported by the current device FW
2. Used checksum [tl_check_fw2.exe](https://github.com/pvvx/ATC_MiThermometer/tree/master/utils)

Old version (up to 2020) FW OTA does not check checksum...
TLSR826x/825x based devices have different FW options. They are distinguished by the location of the OTA-zone in Flash. These differences are relevant for the BLE, MESH, ZigBee FW.

OTA TLSR826x/825x - https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html
FullFlash backup to BLE FW TLSR825x see:  https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker

vevsvevs, 2021-11-26:
Thank you very much for your answer!

Well, it looks like Yeelight remote has OTA disabled (at least it doesn't expose any meaningful UUIDs in pairing mode), which ruins the whole idea ;(

closed 2021-11-26 by vevsvevs