pvvx/ATC_MiThermometer #153: [Improvement] Connection to devices with a name that starts with a number

jhgee, 2021-11-28:
I just noticed that Bluetooth devices starting with a number are filtered out.

When I setup my thermometers I thought it's a good idea to give them a name like `<no>_<MAC>`, e.g. `01_6E8A31`.
Now I just wondered why I can no longer find the device when I click connect. I just saw in the source code of the HTML pages that devices that do not start with a letter are filtered out.

I would suggest to either remove the filtering of devices and make it optional or add a hint when changing the device's name to something starting with a number. These won't appear again in the filtered list after clicking connect.

Thanks for this nice software and all the effort!

pvvx, 2021-11-29:
Hundreds of devices drop out in my city house without this filter ...

closed 2021-12-02 by pvvx

jhgee, 2021-12-02:
Thanks a lot.
I understand, maybe adding this filter option via tick mark could be an option then ;-)