pvvx/ATC_MiThermometer #155: B1.9 Advertising Problem

xfactory9025, 2021-11-30:
My device hardware B1.9. I installed ATC_V34d. The image is as seen in the picture. Temperature and humidity information shows as no value on the phone.
[image]

pvvx, 2021-11-30:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
Log: Hardware Version: LYWSD03MMC B1.9, Software Version: 3.4, Sensor: SHT4x ?

xfactory9025, 2021-11-30:
> https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html Log: Hardware Version: LYWSD03MMC B1.9, Software Version: 3.4, Sensor: SHT4x ?

Yes

pvvx, 2021-11-30:
The reasons for this behavior have not been established. More suitable option - LCD from version 1.4, LCD driver from 1.9?
Could you clarify the production date on the box?

xfactory9025, 2021-12-01:
[image]

Protoncek, 2021-12-01:
I just received two of those (2021.08) and they have B1.9 (i checked by opening them). Flashed both of them with pvvx succesfully.

xfactory9025, 2021-12-01:
> Az iki önce tane (2021.08) aldım ve B1.9'a sahipler (açarak kontrol ettim). Her ikisi de etkili bir şekilde pvx ile flashladım.

I did not understand why this happened.

pvvx, 2021-12-02:
Xiaomi LYWSD03MMC Original Firmware v1.0.0_0130 working fine?

xfactory9025, 2021-12-02:
> Xiaomi LYWSD03MMC Original Firmware v1.0.0_0130 working fine?

yes

pvvx, 2021-12-03:
What version of HW does the original firmware show?

xfactory9025, 2021-12-10:
B1.9

xfactory9025, 2021-12-13:
I have installed atc1441 version. The values ​​do not appear in the advertisement broadcast on the phone, but when used with the esp32 code here, the values ​​appear on the serial monitor. Working with Esp32 solved my problem.

closed 2022-05-01 by pvvx