pvvx/ATC_MiThermometer #156: Maximum number of BLE devices?

Protoncek, 2021-12-03:
Recently i connected some new xiaomi’s, so currently i have totally 8 of them. I start to wonder, is it any limit in number of BLE devices so that they won’t disturb each other? 
Should i change advertising interval on each for just a little, so if two of them “decide” to send it’s data simultaneously they will sooner or later go “out of sync”…? I don’t know BT protocol, so i don’t know how these devices deal with colisions.

I have them connected to two esp32 (esphome) modules, 4 of them work ok for now on each module. Will be ok if i add even more of them to a single esp32?

pvvx, 2021-12-04:
A common sniffer on one channel receives BLE ads from hundreds of devices per second. The bit rate is 1 Mbps. Those. up to 100 kilobytes per second.
The device uses transmission duplication. With default settings, the thermometer provides duplicate readings 4 times in 3 gears on 3 channels.

Protoncek, 2021-12-04:
Aha… many thanks for detailed explanation. So, in short: it should work without any problem…
I used to have over 10 wifi modules (espxx), now i replaced many of them with xiaomi’s (basically most of those which only measure temp and humidity). Main advantage of xiaomi’s is battery power, so if they will work ok i’ll buy some more of them, so it’s good to know that they will work ok.
Thanks again!

pvvx, 2021-12-24:
[image]

pvvx, 2022-03-08:
[image]
There are no gaps in the graphs in HA...
All automation scripts work flawlessly.

Protoncek, 2022-03-08:
Nice! Thanks for info! I bet you already wrote, but i can't remember: which BT receiver system (hardware, FW) do you use?

pvvx, 2022-03-08:
The cheapest USB-BT. Ver Bluetooth 4.0 :) 
```
lsusb -v -s 006:002

Bus 006 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x0a12 Cambridge Silicon Radio, Ltd
  idProduct          0x0001 Bluetooth Dongle (HCI mode)
  bcdDevice           88.91
  iManufacturer           0
  iProduct                2 CSR8510 A10
....
```

Protoncek, 2022-03-08:
Aha...thanks. Since my synology  (where my HA is) is located at one corner of my house using BT dongle wouldn't reach to all my Xiaomi's, that''s why i use ESP32. But, as said a few times already, Tasmota works pretty well, way better than ESPHome.

pvvx, 2022-03-08:
This is in a separate house-workshop for a hobby, 6x6 meters, all internal partitions are wooden...

closed 2023-02-18 by pvvx