pvvx/ATC_MiThermometer #157: Humidity offset / discrepancy

oleg0, 2021-12-04:
Hi Victor,

I've "upgraded" several LYWSD03MMC (HW B1.4) from original FW to the ATC_V34d.bin (thank you for the work!).
With each upgrade I've noticed that humidity readings after upgrade (both on the screen and in BT packets) decreased by ~5% (e.g. read 39% before upgrade and 34% after upgrade). The temperature readings did not change noticeably. The corrections / offsets on TelinkMiFlasher.html are all zeros (never changed from defaults). Such discrepancy was not short term, it remained. 

[image]
Here are two devices in the same room. First (green) got upgraded around 12/02 mark, second (yellow) upgraded around 12/04.

Is there any known reason / issue? Please let me know if I can collect any info to help troubleshooting this.

Thank you
---oleg

pvvx, 2021-12-05:
The official firmware overestimates the readings by + 5%.

oleg0, 2021-12-05:
Thank you for the prompt response and references!

closed 2021-12-05 by oleg0

phoberus, 2022-05-11:
Is this 5% increase justified? So most of these sensors really measure about 5% too low?
I have a second sensor from another manufacturer and it reads 8% lower than the LYWSD03MMC with original firmware.

pvvx, 2022-05-23:
Humidity sensors are subject to degradation. Depends on operating conditions and operating time.