pvvx/ATC_MiThermometer #158: The extending sensor's wiring

almostlunatic, 2021-12-05:
I have several water boilers in my home. They have mechanical thermometers on them as always. I would like to control the average temperature of the water in the boilers remotely and more precisely on the room. Could I extend the sensor's 4 wires to install the sensor in (on) the boiler for seeing the water temperature on the display and on the Domoticz? It's a ready device, why should one ivent a new one on base of something else? Could someone give me any suggestions (maybe some photos) how could it be implemented? 

Thanks in advance!

pvvx, 2021-12-07:
The bus connection to the sensor is I2C. I2C is not designed for long cable connections, but for short PCB connections.

almostlunatic, 2021-12-07:
God point! But rewriting the FW is not my cup of tea. Shure, ite would be nice to have 2 exchanging temp displays: the water boiler's on and the room's on. But I'm not capable to achieve this.

jddonovan, 2022-01-02:
@almostlunatic  Was this a combined boiler or just a hot water one?

I've solved my heating problems without use of wires at my parent's place. I have multiple Mi-sensors that are then picked up by an Atomic Pi with a BT2MQTT bridge. This is then picked up by an ESP that controls the heating. You can choose the chosen heating area and temperature on the web interface or via Home Assistant.

The ESP it self could also pick up BT transmissions if the MQTT server or Wifi goes down. I wouldn't fully trust Home Assistant, as it tends to crash between version numbers.

The boiler is actually controlled by [EMS-ESP](https://emsesp.github.io/). The other ESP merely sends a message to this EMS-EPS to either increase or lower the temperature. The EMS-ESP creates its own wifi hotspot -> no wifi strictly required.

Personally, I would personally do it in the following way: Have one device controlling the boiler and accepting API calls over wifi, BT or ESP-NOW. Then the same device (or another, if the house is big) that collect the temperature data over the house and has the logic inside. And third, a number of these useful little sensors.

If it's JUST the hot tap water for shower and bath, I would personally just strap the BT against the boiler where the insulation is thinnest, close to the water intake valve and cover it with insulation. You can then strap a low current LED on the reset pin and use the temperature trigger to show a light if the temperature goes down too low to alert the users.

If you really want to go the I2C route, the HW versions 1.4 and 1.9(?) have the SDA SCL pins exposed, so you can solder some wires there. E.g. HW 1.8 is missing those. 15-30 cm cables might actually work, so it's not a lost cause. It just makes the sensor placing a challenge. In this case you'd need a sensor which is easy to solder pads and without LDOs etc or remove them before soldering. Reusing the one onboard might be tricky due to tiny size, but doable.

I don't have any actually useful pics, but this HA setup I made is vaguely relevant, if room heating is involved:
[image]

closed 2022-05-01 by pvvx