pvvx/ATC_MiThermometer #160: Why putting MAC into ADV's service data?

danergo, 2021-12-07:
Hi, Folks,

This might be already asked, but I couldn't found: in custom advertisement format for "LYWSD03MMC":

> UUID 0x181A - size 19: Custom format (all data little-endian)

Why do you send the MAC in this data? As the MAC must be already presented in the advertisement packet "de facto". It might not be a huge performance or consumption-saver if we skip these 6 bytes, but sill, it feels unnecessary and I don't get the reason of it.

And another thing: this firmware is pretty impressive, also there are a lot of documentation and resources. 

One thing is not clear though: If I selected *Encrypted beacon*:

[image]

Tooltip says "Bindkey encrypted beacon".

However, down below I can see two options: *Mi Bind Key* and *EEP BindKey*:

[image]

Which one will be used to encrypt the readings?

And is this still the tool to decrypt the reading: https://github.com/pvvx/ATC_MiThermometer/blob/master/test_adv_cust.py ?

Thank you!

pvvx, 2021-12-07:
Bindkey fetch uses priority:
Writing to Flash from the original firmware. If the entry is missing or erased, then the key is taken from the EEP.

The MAC address is not available for user requests on some operating systems. For example, in the js API  for Chrome...

danergo, 2021-12-07:
> The MAC address is not available for user requests on some operating systems. For example, in the js API for Chrome...

Missing piece found, thanks :)

> Writing to Flash from the original firmware. If the entry is missing or erased, then the key is taken from the EEP.

Thank this too. So if I don't erase anything, I will be able to decrypt the data with my actual "Mi Bind Key"?

pvvx, 2021-12-07:
No. When flashing, bindkey was changed. It is necessary to restore to the original, from Mi-Home: "Swap previous token + bindkey".

danergo, 2021-12-07:
Wait a bit. I don't use Mi Home.

I want to decrypt with a python code on Debian Linux. No Mi-Home is involved.

So in case I use your script: https://github.com/pvvx/ATC_MiThermometer/blob/master/test_adv_cust.py

Which bindkey should I use? Should I still need to swap to the previous?

pvvx, 2021-12-07:
Enter "EEP bundkey" and pick "Erase all Mi Keys".

closed 2021-12-07 by danergo