pvvx/ATC_MiThermometer #163: Too low humidity value compared to stock version and calibration result. Maximum humidity offset is only 12.7%.

mookkari, 2021-12-11:
I just had 10 pcs new LYWSD03MMC B1.9 thermometers. All of those have very similar humidity and temperature values in same placement with stock firmware. I flashed most of those sensors one by one to with latest firmware version 3.4. Always after flashing humidity value was decreased instantly by 11-12%. This is only concerning humidity, I do not see any relevant change in temperature value after firmware update.

I have had now all 10 thermometers in Sodium Chloride calibration box about >12 hours and thermometers with stock firmware shows 71-72% that is very close. All thermometers with pvvx firmware shows about 60%. Correct value for calibration should be 75.4%. Meaning that  stock firmware is only 3-4% too low compared to calibration result (after 12h, I will still wait for >24h for final result), but pvvx firmware shows about 15% too low humidity value. I am using this procedure for humidity calibration:
https://www.vernier.com/til/4337
https://www.youtube.com/watch?v=aeLPRZQTA4w

I could use humidity offset to calibrate sensors but firmware config does now allow to set bigger values than 12.7%. If bigger value is given it will be set as 12.7%.

Also temperature and humidity values are not rounded in LCD or in advertisement (using Atc1441 because less digits). If temperature shown in Telink flasher is 23.87 it shows 23.8 in LCD and advertisement. Value 23.87 should be rounded to 23.9.

pvvx, 2021-12-12:
Humidity readings from different meters in the same house:
[image]

pvvx, 2021-12-12:
LYWSD03MMC B1.9 uses SHT4x sensors:
https://www.sensirion.com/en/environmental-sensors/humidity-sensors/humidity-sensor-sht4x
_The SHT4x builds on a completely new and optimized CMOSens® chip that offers reduced power consumption and improved accuracy specifications. With the extended supply voltage range from 1.08 V to 3.6 V, it’s the perfect fit for mobile and battery-driven applications.
The high-accuracy versions SHT41 and SHT45 feature typical and maximal accuracies that have been honed even further, down to ∆RH = ± 1% RH and ∆T = ± 0.1°C, making them suitable for the most demanding applications in all industries._ 

Always after flashing custom FW, the humidity value instantly decreases by 5.0% and begins to meet the of other sensors from other manufacturers.

pvvx, 2021-12-12:
https://github.com/pvvx/ATC_MiThermometer/commit/6a1831b173505c7f7916cca4a41e9084c7077dd3

closed 2021-12-12 by pvvx

mookkari, 2021-12-12:
Thanks! I have now flashed most of my B1.9 thermometers with ATC_V35.bin. Seems that it works and now humidity value is same as with in stock firmware. Also values has been balanced after yesterday and they are very close to same humidity values that AirThings Wave Plus shows. 

I have also few LYWSD03MMC B1.4 with Sensor SHTC3 (also flashed to v3.5). They all show about 5 RH% lower values than B1.9. Is this sensor related or SW issue?

There is workaround for humidity rounding, if I set Humidity offset=0.5 it will do rounding correctly LCD and advertisement when 0 digits used for humidity. This workaround does not work for temperature with 1 digit as Temperature offset=0.05 is not allowed.

pvvx, 2021-12-12:
LYWSD03MMC B1.4 uses SHTV3 sensor. There is no documentation for it, but the commands is close to SHTC3.

xairam, 2022-01-09:
can you tell me where did you get the 1.9? I cant tell the difference on shops selleing them.