pvvx/ATC_MiThermometer #165: Battery consumption

paralichko, 2021-12-15:
Hello,

i have flashed some of my Xiaomi's with this firmware, but i notice high battery consumption. I have reed Your explanation, that this firmware should consume less than the original, but is it possible, the function of reading P8 and the function with the RESET pin trigger to increase the battery usage. Look into the attached picture - the batteries are the same on all devices. The ATC devices are used of max of 7 days. The first Xiaomi ( with the orig firmware ) was used more than 2 months...
[image]

Protoncek, 2021-12-15:
The catch is that original FW doesn't report actual battery state - it always shows as 100% until battery is (practically) emtpy, then immediately shows "low bat". That's why your originals are still at 100%.

paralichko, 2021-12-15:
Hello,

thanks for the answer. Is it the same with the battery voltage ? Is it reported wrongly by the original firmware ?

Protoncek, 2021-12-15:
Huh... i think that voltage is not reported at all in original FW, only "percentage" - more or less in binary form :-) as "good" and "bad".
Since i use all my Xiaomi's in Home Assistant i have them all flashed with pvvx FW. So far all 8 of them work flawlessly.

paralichko, 2021-12-15:
In the home assistant there is battery voltage showing 3.1V all the time. What real batteey life You achive ? 1 month, 6 months, 1 year ?

Protoncek, 2021-12-15:
Couldn't say because i don't have them for long - appr. 2-3 months now, they all show around 80-87%.

For longer battery life i think that you can set a bit longer pause between BT transmits, but even with default setting i've read that guys do have over 6 months of work, even up to a year (i guess that depends on battery quality and age).  I guess that even 6 months is quite excellent for such device.

paralichko, 2021-12-15:
As You have explain, longer than 5sec is not OK? I have tried with 10s, and have lost an hour in recoverihg with reading all the topics and using nRF connect...

Protoncek, 2021-12-15:
I think that  BLE protocol has some limitations regarding publishing interval, yes. I can’t tell details, though, but i do remember something aróund 5 seconds, yes…
Reset  pin can be in only one state if you want device to work. Just let it be, state “as is” when device works is the correct one. 
P8 (if you mean gpio input) is supposely very low current, but i didn’t go into details regarding that.

paralichko, 2021-12-15:
For the reset pin, i mean the function that sets it to 1 or 0 depending on temp/hum

Protoncek, 2021-12-15:
huh... i can't help you here...sorry...

pvvx, 2021-12-15:
Since 12.2020, 3 pieces of thermometers are working on the alternative firmware.
On the two included on 12.2020, but with the original firmware, I have already changed the batteries.

Protoncek, 2021-12-15:
From your graph i assume that these sensors actually CAN be mounted outside in the cold? With original FW they are supposely from 0 to 60 degrees... ? How low will go with your FW?

pvvx, 2021-12-15:
CR2032 Drop from +20 to -10C on average 0.009V per degree ...
The full power supply functionality of the thermometer is limited to about 2V. The TLSR8251 SoC itself will work up to 1.8V (not tested at -40C), but the consumption is pulsed, the CR2032 battery has a large internal resistance and at the peak moment of transmission it will drain ... Writing to the internal Flash is more modest, but there is hope that there will be more efficient for -30C at 2.0V.
Those. The temperature for turning off the thermometer with the data of the battery samples will be about -41C.

---
Submitted to the forum: https://esp8266.ru/forum/threads/tlsr8251-lcd-termometr-lywsd03mmc-xiaomi-bluetooth-termometr.5263/post-83707
Original firmware:
[image]

pvvx, 2021-12-15:
[image]
[image]
[image]

paralichko, 2021-12-15:
How You measure the % of the battery ? It's some kind of coltage to % scale ?

pvvx, 2021-12-16:
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/battery.c#L80
battery_level = (battery_mv - 2200) / ((3100 - 2200) / 100);

50494554524F, 2021-12-23:
i'm using 10000ms adv interval and x6, not encrypted (safe distance from other buildings). 
cheap pkcell batteries have been running for 7 months. 
the voltage is still 2.5-2.6v, ambient temperature 19°C winter -30°C summer... one year goal is possible with cheap batteries too

closed 2023-02-18 by pvvx