pvvx/ATC_MiThermometer #166: Connection issues after flashing

TFMvdBroek, 2021-12-20:
So I've flashed the v3.5 for the Xiaomi Mijia thermometer. It all seemed to go fine.
After flashing no settings showed up, and the flashing software showed that alternative firmware was detected.

I decided to go back to original firmware, so I flashed the one for B1.4/B1.9. This also worked but now I can't connect anymore to flash the custom firmware once again.

I'm getting this error:
`NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
`

Display etc seem to work fine still.

TFMvdBroek, 2021-12-20:
Flashing from my phone seems to work, strange ...

TFMvdBroek, 2021-12-25:
Do you think I can flash others? I ordered 4, they all seem to be hardware B2.0.

Xploder, 2022-01-02:
Mine are B2.0 aswell.
Can we help with info to get those flashed?

Xploder, 2022-01-02:
I was able to dump the flash of my device.
Imagine my disappointment when it checksummed to be the same as "Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130".
[xiaomi_LYWSD03MMC_v1.0.0_0130_B2.0_orig.zip](https://github.com/pvvx/ATC_MiThermometer/files/7799584/xiaomi_LYWSD03MMC_v1.0.0_0130_B2.0_orig.zip)

pvvx, 2022-01-03:
It is necessary to find the differences, to determine the version 2.0 for this firmware.
I don't have version 2.0 yet.

Xploder, 2022-01-03:
Stock Firmware is the same and your version 3.5a works well on B2.0.
If you need any hardware info about B2.0 I'm glad to help.

backerman, 2022-01-08:
3.5 works fine on my B2.0 flashed from Edgium 98.0.1108.13 on Windows 11 build 22000.376.

abarbarich, 2022-01-14:
Confirming I have 2 x B2.0 flashed with 3.5 and Working well, however Device reports that it is B1.7

OscarHanzely, 2022-01-27:
I cannot connect to the device that originally reported B2.0 and later reported B1.7 anymore. The 3.5c did not seem to work and now I have the exact same issue as the main poster. Sorry, I did't see that and opened new issue. It could be closed as a duplicate.

Basicaly after flashing with custom firmware the device could not be connected to.

pvvx, 2022-01-28:
This will most likely help you:
https://github.com/pvvx/ATC_MiThermometer/issues/172#issuecomment-1009670452

mjsage, 2022-11-17:
I also received a B2.0 version today, flashed it from android with the current version 3.8 and after that it kept disconnecting shortly after connection. 

Connected just long enough to start it flashing back to original firmware, and uploading the firmware seemed to keep it connected until complete.

nikito7, 2022-11-17:
Same here. Back to original and then 3.8 work

mjsage, 2022-11-18:
> I received a B2.0 version today, flashed it from android with the current version 3.8 and after that it kept disconnecting shortly after connection.
> 
> Connected just long enough to start it flashing back to original firmware, and uploading the firmware seemed to keep it connected until complete.

Meant to add that putting a brand new battery in did seem to give a very tiny bit longer before some disconnects, which helped to make it possible to start the flash back to oem firmware. The one I used originally had been briefly used to test something and was down to just over 3V. 

With the oem firmware restored, a connection to telinkmiflasher still dropped once in a while, but after setting it up in the OEM app, the connection to telink seemed solid.  Something in the stored config perhaps?

Also after flashing the 3.8 firmware on, the flasher identified the device as B1.9 instead of B2.0 it had previously.

mjsage, 2022-11-18:
> Same here. Back to original and then 3.8 work

In that case I might try again then and if it connects long enough to change anything, try changing the connect latency as well  https://github.com/pvvx/ATC_MiThermometer/issues/11

and maybe add a cap.

closed 2023-02-18 by pvvx