pvvx/ATC_MiThermometer #167: LYWSD03MMC:  Sensor values not updated below -12'C-ish

jjwbruijn, 2021-12-21:
Hi, thanks for your awesome work on this! I have an issue with LYWSD03MMC thermometers placed outside, in my shed and in the garden. I'm using an ESP32 to pick up advertisements in 'custom' mode. The sensors are running V35a.

It seems the thermometer will not go lower than about -12'C. It keeps advertising -11.41 with the same battery and humidity, with a real temperature of about -25. The display will not show lower than about -12'C either.. The SHTC3 sensor should be able to go a lot lower, but there seems to be something preventing correct operation at lower temperatures.

```
main/sensor/A4C1385AAAAA/state {"temperature":-1.19,"humidity":79.75,"battery":86}
main/sensor/A4C1385AAAAA/state {"temperature":-2.08,"humidity":78.06,"battery":85}
main/sensor/A4C1385AAAAA/state {"temperature":-2.08,"humidity":78.06,"battery":85}
main/sensor/A4C1385AAAAA/state {"temperature":-2.49,"humidity":77.01,"battery":84}
main/sensor/A4C1385AAAAA/state {"temperature":-2.49,"humidity":77.01,"battery":84}
main/sensor/A4C1385AAAAA/state {"temperature":-2.89,"humidity":75.96,"battery":84}
main/sensor/A4C1385AAAAA/state {"temperature":-2.89,"humidity":75.96,"battery":84}
main/sensor/A4C1385AAAAA/state {"temperature":-2.89,"humidity":75.96,"battery":84}
main/sensor/A4C1385AAAAA/state {"temperature":-2.89,"humidity":75.96,"battery":84}
main/sensor/A4C1385AAAAA/state {"temperature":-3.31,"humidity":74.65,"battery":83}
main/sensor/A4C1385AAAAA/state {"temperature":-3.31,"humidity":74.65,"battery":83}
main/sensor/A4C1385AAAAA/state {"temperature":-3.31,"humidity":74.65,"battery":83}
main/sensor/A4C1385AAAAA/state {"temperature":-3.7,"humidity":72.18,"battery":82}
main/sensor/A4C1385AAAAA/state {"temperature":-3.7,"humidity":72.18,"battery":82}
main/sensor/A4C1385AAAAA/state {"temperature":-3.7,"humidity":72.18,"battery":82}
main/sensor/A4C1385AAAAA/state {"temperature":-4.09,"humidity":68.81,"battery":82}
main/sensor/A4C1385AAAAA/state {"temperature":-4.09,"humidity":68.81,"battery":82}
main/sensor/A4C1385AAAAA/state {"temperature":-4.09,"humidity":68.81,"battery":82}
main/sensor/A4C1385AAAAA/state {"temperature":-4.47,"humidity":65.54,"battery":81}
main/sensor/A4C1385AAAAA/state {"temperature":-4.47,"humidity":65.54,"battery":81}
main/sensor/A4C1385AAAAA/state {"temperature":-4.47,"humidity":65.54,"battery":81}
main/sensor/A4C1385AAAAA/state {"temperature":-4.88,"humidity":62.78,"battery":81}
main/sensor/A4C1385AAAAA/state {"temperature":-4.88,"humidity":62.78,"battery":81}
main/sensor/A4C1385AAAAA/state {"temperature":-4.88,"humidity":62.78,"battery":81}
main/sensor/A4C1385AAAAA/state {"temperature":-4.88,"humidity":62.78,"battery":81}
main/sensor/A4C1385AAAAA/state {"temperature":-5.29,"humidity":60.29,"battery":80}
main/sensor/A4C1385AAAAA/state {"temperature":-5.29,"humidity":60.29,"battery":80}
main/sensor/A4C1385AAAAA/state {"temperature":-5.29,"humidity":60.29,"battery":80}
main/sensor/A4C1385AAAAA/state {"temperature":-5.58,"humidity":58.22,"battery":80}
main/sensor/A4C1385AAAAA/state {"temperature":-5.58,"humidity":58.22,"battery":80}
main/sensor/A4C1385AAAAA/state {"temperature":-5.58,"humidity":58.22,"battery":80}
main/sensor/A4C1385AAAAA/state {"temperature":-5.94,"humidity":56.5,"battery":79}
main/sensor/A4C1385AAAAA/state {"temperature":-5.94,"humidity":56.5,"battery":79}
main/sensor/A4C1385AAAAA/state {"temperature":-6.35,"humidity":55.13,"battery":79}
main/sensor/A4C1385AAAAA/state {"temperature":-6.35,"humidity":55.13,"battery":79}
main/sensor/A4C1385AAAAA/state {"temperature":-6.67,"humidity":53.94,"battery":78}
main/sensor/A4C1385AAAAA/state {"temperature":-6.67,"humidity":53.94,"battery":78}
main/sensor/A4C1385AAAAA/state {"temperature":-6.67,"humidity":53.94,"battery":78}
main/sensor/A4C1385AAAAA/state {"temperature":-6.67,"humidity":53.94,"battery":78}
main/sensor/A4C1385AAAAA/state {"temperature":-7,"humidity":52.95,"battery":78}
main/sensor/A4C1385AAAAA/state {"temperature":-7.31,"humidity":52.11,"battery":78}
main/sensor/A4C1385AAAAA/state {"temperature":-7.31,"humidity":52.11,"battery":78}
main/sensor/A4C1385AAAAA/state {"temperature":-7.64,"humidity":51.41,"battery":77}
main/sensor/A4C1385AAAAA/state {"temperature":-7.95,"humidity":50.85,"battery":77}
main/sensor/A4C1385AAAAA/state {"temperature":-7.95,"humidity":50.85,"battery":77}
main/sensor/A4C1385AAAAA/state {"temperature":-8.27,"humidity":50.45,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-8.27,"humidity":50.45,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-8.27,"humidity":50.45,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-8.62,"humidity":50.11,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-8.62,"humidity":50.11,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-8.62,"humidity":50.11,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-8.89,"humidity":49.87,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-8.89,"humidity":49.87,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-8.89,"humidity":49.87,"battery":76}
main/sensor/A4C1385AAAAA/state {"temperature":-9.19,"humidity":49.72,"battery":75}
main/sensor/A4C1385AAAAA/state {"temperature":-9.19,"humidity":49.72,"battery":75}
main/sensor/A4C1385AAAAA/state {"temperature":-9.19,"humidity":49.72,"battery":75}
main/sensor/A4C1385AAAAA/state {"temperature":-9.19,"humidity":49.72,"battery":75}
main/sensor/A4C1385AAAAA/state {"temperature":-9.49,"humidity":49.64,"battery":74}
main/sensor/A4C1385AAAAA/state {"temperature":-9.49,"humidity":49.64,"battery":74}
main/sensor/A4C1385AAAAA/state {"temperature":-9.83,"humidity":49.56,"battery":74}
main/sensor/A4C1385AAAAA/state {"temperature":-9.83,"humidity":49.56,"battery":74}
main/sensor/A4C1385AAAAA/state {"temperature":-9.83,"humidity":49.56,"battery":74}
main/sensor/A4C1385AAAAA/state {"temperature":-10.04,"humidity":49.53,"battery":73}
main/sensor/A4C1385AAAAA/state {"temperature":-10.04,"humidity":49.53,"battery":73}
[... 35 lines cut ...]
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
main/sensor/A4C1385AAAAA/state {"temperature":-11.41,"humidity":49.75,"battery":71}
```

pvvx, 2021-12-21:
https://github.com/pvvx/ATC_MiThermometer/issues/165#issuecomment-995238662

jjwbruijn, 2021-12-21:
> [#165 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/165#issuecomment-995238662)
Thanks, interesting limitation of the device. I realize the battery voltage will drop with decreasing temperatures, however, the voltage should still be ample for proper functioning of the device at this point. The advertisements are still being sent, but they're not being updated, for some reason... The sensor should have a minimum voltage of 1.62, a lot lower than the SoC, so that shouldn't be too much of an issue

pvvx, 2021-12-21:
[image]
Temperature on the windowsill. Just measured on B1.4.
Outside temperature.

pvvx, 2021-12-21:
Brought it out into the street. I put it on an iron structure:
[image]

pvvx, 2021-12-22:
[image]

jjwbruijn, 2021-12-22:
Okay, I'll mark it down as 'unable to reproduce' :) Thanks for your graphs. Must be something weird with my batch of thermometers. Can you elaborate on your advice not to use an ESP32?

Protoncek, 2021-12-22:
I'd like to know any alternatives to ESP32, too. BT on Pi is not an option for me, since i have HA on Synology, and also i need more than one BT receiver for all my Xiaomi's.
So, any suggestion would be good. Preferably working on ESPHome..., ok MQTT is an option, too.

jjwbruijn, 2021-12-22:
Due to some specific requirements, I wrote my own BLE-sniffer/gateway using nimBLE on ESP32. It's working fine really. It sends BLE ad's to an MQTT server, after some decoding. I haven't really encountered major problems with this, and the energy consumption is very low. So i'm interested to know what problems pvvx has with it :)

pvvx, 2021-12-22:
> Can you elaborate on your advice not to use an ESP32?

- Lots of ad package skipping (BLE driver).
- No message about lost reception from device (soft).

jjwbruijn, 2021-12-22:
I see your points. While I wouldn't design a commercial product with it, it's been working acceptably for me for some time now. It does get rather warm when receiving advertisement, but consumption is still a lot lower than RaspPi or other linux-based boards, so for now, I'm happy :)

I do wonder if it contains any undisclosed 'bugs' in silicon that may compromise security later on... I have some trust issues with chips from China ;)

jjwbruijn, 2021-12-23:
Started and flashed a new thermometer, from another batch. Seems happy all the way down to 27´C, my fridge wouldn't go any lower. Not gonna waste anyone's time any further, I'm gonna assume this was a hardware thing with two of my older thermometers. Thanks Victor!
[image]

closed 2021-12-23 by jjwbruijn

pvvx, 2021-12-24:
The CR2032 battery only lasts for half the operating time at -25 °C.
Panasonic CR2032:
[image]

https://industrial.panasonic.com/ww/products/pt/lithium-batteries/models