pvvx/ATC_MiThermometer #172: Unable to pair/connect with bluetoothctl

mvdklip, 2022-01-01:
I'm not sure if there's something wrong with the setup of the bluetooth controller in my raspberry pi 3 so I'm seeking for confirmation here that what I'm trying to do should work with custom firmware.

Using bluetoothctl with transport set to 'le' I'm trying to do the following things:

- scan for advertisements - works fine
- trust the mac address - works fine
- pair with the mac address - org.bluez.Error.AuthenticationFailed
- connect with the mac address - org.bluez.Error.Failed

I tried this with multiple LYWSD03MMC and CGG1-M devices and always get above results. I should be able to pair and/or connect to these devices running custom firmware, right?

The reason I'm trying to connect is because I want to send custom data to the LCD. Receiving and decoding advertisements from these devices has always worked fine but that doesn't require an active connection.

I also tried gatttool but I get Transport endpoint is not connected there. Supposedly gatttool is not supported anymore so I'm not sure if that has anything to do with my bluetoothctl issues.

pvvx, 2022-01-03:
Perhaps something from what is described here will help you:
https://github.com/pvvx/ATC_MiThermometer/issues/174#issuecomment-1003987084

mvdklip, 2022-01-03:
Unfortunately no, it doesn't help. I tried LYWSD03MMC with PIN, LYWSD03MMC without PIN and CGG1-M without PIN. It is as if this bluetooth adapter cannot connect/pair at all. It scans fine but refuses to connect. :-(

I will see if I can pair/connect to another type of bluetooth device and/or if I can insert a second (USB) receiver in this raspberry pi.

mvdklip, 2022-01-03:
Got a bit further by using btmon. It seems to create a connection successfully but then cancel it for some reason:

```
< HCI Command: LE Create Connection (0x08|0x000d) plen 25                                    #20 [hci0] 20:54:57.467904
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: White list is not used (0x00)
        Peer address type: Public (0x00)
        Peer address: 58:2D:34:12:30:E1 (Qingping Electronics (Suzhou) Co., Ltd)
        Own address type: Public (0x00)
        Min connection interval: 30.00 msec (0x0018)
        Max connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                    #21 [hci0] 20:54:57.468586
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Create Connection Cancel (0x08|0x000e) plen 0                              #22 [hci0] 20:55:01.620448
> HCI Event: Command Complete (0x0e) plen 4                                                  #23 [hci0] 20:55:01.623559
      LE Create Connection Cancel (0x08|0x000e) ncmd 1
        Status: Success (0x00)
```

Now looking at https://www.spinics.net/lists/linux-bluetooth/msg67241.html if that provides some hint on what's going wrong. Reporting it here in case some else using this firmware runs into the same problem.

pvvx, 2022-01-04:
For the BLE standard, the maximum advertising interval is 10 seconds.

mvdklip, 2022-01-04:
I know. The Linux kernel people basically broke it by implenting this
"auto" connection stuff with its 2s timeout. The workaround is to stay
within 2s advertising interval if you want to be able to connect fron such
a broken kernel.

pvvx, 2022-01-04:
[Core Specification 5.3](https://www.bluetooth.com/specifications/specs/core-specification/)
Bluetooth Core Specification
Bluetooth® Specification
▪ Revision: v5.3
▪ Revision Date: 2021-07-13
▪ Group Prepared By: Core Specification Working Group

4.4.2.2.1 Advertising interval
````
The advertising interval (advInterval) shall be an integer multiple of 0.625 ms in
the range 20 ms to 10,485.759375 s.
````

pvvx, 2022-01-05:
[image]
+ set advertising interval 1 sec
```
[bluetooth]# scan on
Discovery started
[CHG] Controller AA:AA:AA:AA:AA:AA Discovering: yes
...
[NEW] Device A4:C1:38:0B:5E:ED ATC_0B5EED
[bluetooth]# scan off
Discovery stopped
[CHG] Controller AA:AA:AA:AA:AA:AA Discovering: no
[CHG] Device A4:C1:38:0B:5E:ED RSSI is nil
[bluetooth]# pair A4:C1:38:0B:5E:ED
Attempting to pair with A4:C1:38:0B:5E:ED
[CHG] Device A4:C1:38:0B:5E:ED Connected: yes
Request passkey
[agent] Enter passkey (number in 0-999999): ******
[CHG] Device A4:C1:38:0B:5E:ED UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device A4:C1:38:0B:5E:ED UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device A4:C1:38:0B:5E:ED UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device A4:C1:38:0B:5E:ED UUIDs: 0000180f-0000-1000-8000-00805f9b34fb
[CHG] Device A4:C1:38:0B:5E:ED UUIDs: 0000181a-0000-1000-8000-00805f9b34fb
[CHG] Device A4:C1:38:0B:5E:ED UUIDs: 00001f10-0000-1000-8000-00805f9b34fb
[CHG] Device A4:C1:38:0B:5E:ED UUIDs: 00010203-0405-0607-0809-0a0b0c0d1912
[CHG] Device A4:C1:38:0B:5E:ED ServicesResolved: yes
[CHG] Device A4:C1:38:0B:5E:ED Paired: yes
[NEW] Primary Service
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service0008
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service0008/char0009
        00002a05-0000-1000-8000-00805f9b34fb
        Service Changed
[NEW] Descriptor
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service0008/char0009/desc000b
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Primary Service
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service000c
        0000180a-0000-1000-8000-00805f9b34fb
        Device Information
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service000c/char000d
        00002a24-0000-1000-8000-00805f9b34fb
        Model Number String
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service000c/char000f
        00002a25-0000-1000-8000-00805f9b34fb
        Serial Number String
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service000c/char0011
        00002a26-0000-1000-8000-00805f9b34fb
        Firmware Revision String
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service000c/char0013
        00002a27-0000-1000-8000-00805f9b34fb
        Hardware Revision String
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service000c/char0015
        00002a28-0000-1000-8000-00805f9b34fb
        Software Revision String
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service000c/char0017
        00002a29-0000-1000-8000-00805f9b34fb
        Manufacturer Name String
[NEW] Primary Service
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service001d
        0000181a-0000-1000-8000-00805f9b34fb
        Environmental Sensing
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service001d/char001e
        00002a1f-0000-1000-8000-00805f9b34fb
        Unknown
[NEW] Descriptor
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service001d/char001e/desc0020
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service001d/char0021
        00002a6e-0000-1000-8000-00805f9b34fb
        Temperature
[NEW] Descriptor
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service001d/char0021/desc0023
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service001d/char0024
        00002a6f-0000-1000-8000-00805f9b34fb
        Humidity
[NEW] Descriptor
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service001d/char0024/desc0026
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Primary Service
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service0027
        00010203-0405-0607-0809-0a0b0c0d1912
        Vendor specific
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service0027/char0028
        00010203-0405-0607-0809-0a0b0c0d2b12
        Vendor specific
[NEW] Descriptor
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service0027/char0028/desc002a
        00002901-0000-1000-8000-00805f9b34fb
        Characteristic User Description
[NEW] Primary Service
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service002b
        00001f10-0000-1000-8000-00805f9b34fb
        Unknown
[NEW] Characteristic
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service002b/char002c
        00001f1f-0000-1000-8000-00805f9b34fb
        Unknown
[NEW] Descriptor
        /org/bluez/hci0/dev_A4_C1_38_0B_5E_ED/service002b/char002c/desc002e
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
Pairing successful
[ATC_0B5EED]#
```

mvdklip, 2022-01-05:
> So now I'm curious and thinking if it was not bringing them near the computer the thing that helped and was just pure luck and if I should lower the advertising time (which I guess impacts battery life)...

My advertising time was set to 5000ms and that was too high for my kernel to ever connect. It would just not work at all. I then changed to 1000ms which worked wonders ofcourse. In the end I set them to 2500ms (default value) because I was curious if I would then end up being able to connect sometimes (luck). This turned out not to be the case. Even with 2500ms I can connect reliably. With 5000ms it is totally impossible. I tried that. I didn't try all values in between.

> > Solution: It seems that on more recent Linux kernels it's only possible to connect to these kind of BLE devices if the advertising interval is smaller than 2 seconds because of a newly added timeout in the kernel.
> 
> Do you have any references for this? Because I use Arch Linux and I should have a pretty fresh kernel and Bluetooth stack.

Read the stackoverflow thread linked above.

skgsergio, 2022-01-05:
> Now looking at https://www.spinics.net/lists/linux-bluetooth/msg67241.html and https://stackoverflow.com/questions/43529336/bluez-5-unknown-connection-identifier if those provide some hints on what's going wrong. Reporting it here in case some else using this firmware runs into the same problem.
>
> ...
> 
> **Solution: It seems that on more recent Linux kernels it's only possible to connect to these kind of BLE devices if the advertising interval is smaller than 2 seconds because of a newly added timeout in the kernel.**

Ok after reading those references and doing some digging and if I understood correctly:

Kernel <3.17: 20 seconds timeout for BLE connections.
Kernel >=3.17,<4.16: added a timeout of 2 seconds for BLE auto-connections: https://github.com/torvalds/linux/commit/09ae260ba452c2ed36ec295941a58cb75db213ed
Kernel >=4.16: raised previous timeout to 4 seconds: https://github.com/torvalds/linux/commit/1f01d8be0e6a04bd682a55f6d50c14c1679e7571 (https://lists.ubuntu.com/archives/kernel-team/2017-November/088166.html) 

This last change is from a later date than the SO and mailing list posts so I guess that 2500 is indeed good enough and my problem was 100% signal strength.

mvdklip, 2022-01-05:
> Kernel <3.17: 20 seconds timeout for BLE connections. Kernel >=3.17,<4.16: added a timeout of 2 seconds for BLE auto-connections: [torvalds/linux@09ae260](https://github.com/torvalds/linux/commit/09ae260ba452c2ed36ec295941a58cb75db213ed) Kernel >=4.16: raised previous timeout to 4 seconds: [torvalds/linux@1f01d8b](https://github.com/torvalds/linux/commit/1f01d8be0e6a04bd682a55f6d50c14c1679e7571) (https://lists.ubuntu.com/archives/kernel-team/2017-November/088166.html)

Thanks for the digging! That totally explains my situation because I'm running a kernel > 4.16 so apparently the timeout is 4s in my case and that's why 2500ms works but 5000ms doesn't.

The signal strength / connection issues are more apparent in LYWSD03MMC devices because of a missing condensator. You can find more info on that elsewhere. I actually replaced one of my LYWSD03MMC devices by a CGG1-M because I want to use it as a display for external data and I need to reliably be able to connect to it.

pvvx, 2022-01-07:
> Thanks for the digging! That totally explains my situation because I'm running a kernel > 4.16 so apparently the timeout is 4s in my case and that's why 2500ms works but 5000ms doesn't.

The reliability of the connection cannot be at the level of 2500 ms. One lost reception and the second sending of a connection request will be in 5000..5020 ms, when the second advertisement arrives.

pvvx, 2022-01-07:
> Kernel <3.17: 20 seconds timeout for BLE connections.
> Kernel >=3.17,<4.16: added a timeout of 2 seconds for BLE auto-connections: [...]

This means there is no Bluetooth LE support on Linux!

Protoncek, 2022-01-07:
> The signal strength / connection issues are more apparent in LYWSD03MMC devices because of a missing condensator.

Interesting... but this capacitor can easily be added. Is that capacitor between + and - lines of battery? 
On this pic from another issue:
[image]

there are C24 and C25. Which value they should be? I'm guessing that smaller is around 100nF and bigger a few micro...?

pvvx, 2022-01-07:
Desirable more than 10 μF (С24) and more 100 nF (C25)

Bitman-GR, 2022-01-07:
If these capacitors can be omitted in normal operation by Xiaomi  , adding them wouldn't increase standby leakage current ?

Protoncek, 2022-01-07:
pvvx: thanks for info! I have such SMD's at home, so i'll solder them. 
Bitman-GR: no, these SMD capacitors don't have leakage current, so don't worry.

skgsergio, 2022-01-07:
> This means there is no Bluetooth LE support on Linux!
> The standard connection time prescribed as recommended is 20 seconds. This is taking into account that the maximum advertising interval is 10 seconds.

When they first implemented it they set a 20 seconds timeout. But after some time they added the 2 seconds timeout (that later they raised it to 4) for "auto-connections".

Is not clear to me (mostly because I do not have knowledge of the BLE protocol) what they mean by auto-connections, they mention that is when a connection is established as a consequence of receiving an advertising report, but no clue what this really means.

This is the commit where they first included it:

> When we establish connections as a consequence of receiving an
> advertising report it makes no sense to wait the normal 20 second LE
> connection timeout. This patch modifies the hci_connect_le function to
> take an extra timeout value and uses a lower 2 second timeout for the
> auto-connection case. This timeout is intentionally chosen to be just a
> bit higher than the 1.28 second timeout that High Duty Cycle Advertising
> uses.
> -- https://github.com/torvalds/linux/commit/09ae260ba452c2ed36ec295941a58cb75db213ed

pvvx, 2022-01-07:
> Is not clear to me (mostly because I do not have knowledge of the BLE protocol) what they mean by auto-connections, they mention that is when a connection is established as a consequence of receiving an advertising report, but no clue what this really means.

When the device is paired and does not have a keyboard or display, and the pin code is remembered, a quick connection with negotiation is performed.

pvvx, 2022-01-09:
nRF Connect, Advertiser, set option: Connectable, interval: 4000 (2500 ms):
[image]
```
[bluetooth]# scan on
Discovery started
[NEW] Device 5D:26:71:A1:D7:D4 Redmi Note 7
[bluetooth]# scan off
[bluetooth]# connect 5D:26:71:A1:D7:D4
Attempting to connect to 5D:26:71:A1:D7:D4
Failed to connect: org.bluez.Error.Failed
```

skgsergio, 2022-01-09:
@pvvx I've replicated your test with the same nRF Connect parameters and with this kernel and Bluetooth stack versions:

```
sconde@puck ~ $ bluetoothctl --version
bluetoothctl: 5.62

sconde@puck ~ $ uname -a               
Linux puck 5.15.13-arch1-1 #1 SMP PREEMPT Wed, 05 Jan 2022 16:20:59 +0000 x86_64 GNU/Linux
```

I can connect with reliability using interval 4000 (2500ms), I've made several tests.

```
[bluetooth]# menu scan
[bluetooth]# transport le
[bluetooth]# pattern Mi9TPro
[bluetooth]# back

[bluetooth]# scan on
[NEW] Device 5F:4E:AA:CA:C4:1B Mi9TPro
[CHG] Device 5F:4E:AA:CA:C4:1B RSSI: -75

[bluetooth]# scan off
Discovery stopped
[CHG] Device 5F:4E:AA:CA:C4:1B RSSI is nil
[CHG] Controller 94:B8:6D:87:6C:A6 Discovering: no

[bluetooth]# connect 5F:4E:AA:CA:C4:1B 
Attempting to connect to 5F:4E:AA:CA:C4:1B
[CHG] Device 5F:4E:AA:CA:C4:1B Connected: yes
Connection successful
[NEW] Primary Service (Handle 0x8061)
	/org/bluez/hci0/dev_5F_4E_AA_CA_C4_1B/service0001
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[NEW] Characteristic (Handle 0x4ece)
	/org/bluez/hci0/dev_5F_4E_AA_CA_C4_1B/service0001/char0002
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
[CHG] Device 5F:4E:AA:CA:C4:1B UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 5F:4E:AA:CA:C4:1B UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 5F:4E:AA:CA:C4:1B ServicesResolved: yes

[Mi9TPro]# disconnect 5F:4E:AA:CA:C4:1B 
Attempting to disconnect from 5F:4E:AA:CA:C4:1B
[CHG] Device 5F:4E:AA:CA:C4:1B ServicesResolved: no
Successful disconnected
[CHG] Device 5F:4E:AA:CA:C4:1B Connected: no

[bluetooth]# connect 5F:4E:AA:CA:C4:1B 
Attempting to connect to 5F:4E:AA:CA:C4:1B
[CHG] Device 5F:4E:AA:CA:C4:1B Connected: yes
Connection successful
[CHG] Device 5F:4E:AA:CA:C4:1B ServicesResolved: yes

[Mi9TPro]# disconnect 5F:4E:AA:CA:C4:1B 
Attempting to disconnect from 5F:4E:AA:CA:C4:1B
[CHG] Device 5F:4E:AA:CA:C4:1B ServicesResolved: no
Successful disconnected
[CHG] Device 5F:4E:AA:CA:C4:1B Connected: no
```

However, if I change the interval to 8000 (5000ms) it is impossible to connect!

pvvx, 2022-01-09:
> I can connect with reliability using interval 4000 (2500ms), I've made several tests.
> However, if I change the interval to 8000 (5000ms) it is impossible to connect!

Sometimes it connects to a thermometer and a smartphone with a timing of 2.5 seconds. But more often than not, it's a mistake.

Even on NanoPi R1, OpenWrt 18.06.1:
```
root@FriendlyWrt:~# bluetoothctl --version
bluetoothctl: 5.49
root@FriendlyWrt:~# uname -a
Linux FriendlyWrt 4.14.111 #1 SMP Sat Dec 28 01:30:07 CST 2019 armv7l GNU/Linux
````

skgsergio, 2022-01-09:
> Sometimes it connects to a thermometer and a smartphone with a timing of 2.5 seconds. But more often than not, it's a mistake.
>
> Even on NanoPi R1, OpenWrt 18.06.1:
> ```
> root@FriendlyWrt:~# [...]

That kernel version has the timeout set to 2s, that's probably why. 

<3.17 - No specific limit (20s).
>=3.17, <4.16 - Specific limit of 2s.
>4.16 - Specific limit raised to 4s.

I'll recompile the kernel with a higher limit and perform some tests, and will ask why this limit in the mailing list.

skgsergio, 2022-01-09:
I recompiled my kernel version with the following patch (setting the timeout to 20 seconds) and I can connect setting nRF Connect with interval to 16000 (10000ms), it still fails sometimes. In any case, I didn't read the specification for the recommended timeout, but I guess is max interval x2 so maybe with 21s is more reliable.

With lower intervals, I tested 12000 (7500ms), and 8000 (5000ms), connected every time. So... I guess I'll be asking in the Bluetooth kernel mailing list tomorrow.

```diff
diff --git a/include/net/bluetooth/hci.h b/include/net/bluetooth/hci.h
index b80415011dcd..562c27b12d40 100644
--- a/include/net/bluetooth/hci.h
+++ b/include/net/bluetooth/hci.h
@@ -344,7 +344,7 @@ enum {
 #define HCI_AUTO_OFF_TIMEOUT	msecs_to_jiffies(2000)	/* 2 seconds */
 #define HCI_POWER_OFF_TIMEOUT	msecs_to_jiffies(5000)	/* 5 seconds */
 #define HCI_LE_CONN_TIMEOUT	msecs_to_jiffies(20000)	/* 20 seconds */
-#define HCI_LE_AUTOCONN_TIMEOUT	msecs_to_jiffies(4000)	/* 4 seconds */
+#define HCI_LE_AUTOCONN_TIMEOUT	msecs_to_jiffies(20000)	/* 20 seconds */
 
 /* HCI data types */
 #define HCI_COMMAND_PKT		0x01
```

pvvx, 2022-01-10:
@skgsergio - Useless venture. Linux is long dead - it was killed by the Arduino generation.
There are hundreds of results and conferences on the topic about the small "HCI_LE_AUTOCONN_TIMEOUT" in Google, and there will not be any fixes.

pvvx, 2022-01-10:
Set ATC_0B5EED advertise interval to 10000 ms.
Initialize NanoPi R1...
```
OpenWrt 18.06.1, r7258-5eb055306f
root@FriendlyWrt:~# hciattach /dev/ttyS3 bcm43xx 1500000
bcm43xx_init
Patch not found for BCM43430A1, continue anyway
Set Controller UART speed to 1500000 bit/s
Device setup complete
root@FriendlyWrt:~# hciconfig hci0 up
```
Test 'bluetoothctl'
```
root@FriendlyWrt:~# bluetoothctl
Agent registered
[bluetooth]# scan on
Discovery started
.......................
….. none A4:C1:38:0B:5E:ED ATC_0B5EED (!)
[bluetooth]# devices
Device xx:xx:xx:xx:xx:xx Qingping Motion & Light
Device xx:xx:xx:xx:xx:xx Qingping Motion & Light
Device xx:xx:xx:xx:xx:xx LYWSD02
[bluetooth]# exit
```
No ATC_0B5EED scan for 10 minutes!
Let's see what 'hcitool' says in the scan:
```
root@FriendlyWrt:~# hcitool lescan
LE Scan ...
xx:xx:xx:xx:xx:xx ATC_05F8FA
xx:xx:xx:xx:xx:xx ATC_053CF7
xx:xx:xx:xx:xx:xx Qingping Motion & Light
xx:xx:xx:xx:xx:xx LYWSD02
xx:xx:xx:xx:xx:xx Qingping Motion & Light
…
A4:C1:38:0B:5E:ED ATC_0B5EED
```
Scan ATC_0B5EED finds at once!
What's in 'bluetoothctl'?
```
root@FriendlyWrt:~# bluetoothctl
Agent registered
[bluetooth]# devices
Device xx:xx:xx:xx:xx:xx LYWSD02
Device xx:xx:xx:xx:xx:xx Qingping Motion & Light
Device xx:xx:xx:xx:xx:xx Qingping Motion & Light
[bluetooth]# exit
```
None device ATC_0B5EED !
Go 'hcitool' and connect... (you can do it right away, without any scans.)
```
root@FriendlyWrt:~# hcitool lecc A4:C1:38:0B:5E:ED
Connection handle 64
```
Connection!

Go 'bluetoothctl'...
```
root@FriendlyWrt:~# bluetoothctl
Agent registered
[ATC_0B5EED]# info
Device A4:C1:38:0B:5E:ED (public)
        Name: ATC_0B5EED
        Alias: ATC_0B5EED
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Environmental Sensing     (0000181a-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (00001f10-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00010203-0405-0607-0809-0a0b0c0d1912)
[ATC_0B5EED]#
```
Linux, what can I say ... :) :)

pvvx, 2022-01-10:
Windows 10, Chrome and BT-USB adapters.
The device has an Advertising Interval of 10 sec.

Adapter labeled 'Basseus' - scanning ok.
Adapter marked "B 5.0" - scanning does not find the device. With a shorter interval - approx.
...

skgsergio, 2022-01-10:
I wrote to the mailing list (no response yet, but I've found later an old 2016 thread where they say what is their reason not to increase it: [link to the response](https://lore.kernel.org/linux-bluetooth/9942937A-4799-4AA2-9551-8FEBF110550B@holtmann.org/)).

Anyway, then I found digging more in the kernel and in bluez that there is a Bluetooth management API and it allows you to change the timeout.

The downside is that is only present since kernel 5.9 and bluez 5.56:
- Kernel commit: https://github.com/torvalds/linux/commit/49b020c1d236a36a4533e7db6d2604cb57ed4c51
- Bluez commit: https://github.com/bluez/bluez/commit/86ff0aec4ceb57939bbf6a555468dd2b43332584

This can be called using bluez's `btmgmt` tool. The command is `set-sysconfig` and the option needed is `001b:2:xxxx` where `xxxx` is the timeout in hex little-endian.

For example, with nRF Connect setting interval to 12000 (7500ms):
```
sconde@puck ~/tmp/bluez $ printf "%04x\n" 20000
4e20
sconde@puck ~/tmp/bluez $ sudo btmgmt set-sysconfig -v 001b:2:204e
Set default system configuration success

sconde@puck ~/tmp/bluez $ bluetoothctl
[bluetooth]# menu scan
[bluetooth]# transport le
[bluetooth]# pattern Mi9TPro
[bluetooth]# back

[bluetooth]# scan on
Discovery started
[CHG] Controller 94:B8:6D:87:6C:A6 Discovering: yes
[NEW] Device 63:4B:74:81:8F:BB Mi9TPro

[bluetooth]# connect 63:4B:74:81:8F:BB
Attempting to connect to 63:4B:74:81:8F:BB
[CHG] Device 63:4B:74:81:8F:BB RSSI: -35
[CHG] Device 63:4B:74:81:8F:BB Connected: yes
Connection successful
[NEW] Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_63_4B_74_81_8F_BB/service0001
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_63_4B_74_81_8F_BB/service0001/char0002
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
[CHG] Device 63:4B:74:81:8F:BB UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 63:4B:74:81:8F:BB UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 63:4B:74:81:8F:BB ServicesResolved: yes
```

It is pretty hidden but it worked for me, it has the same effect and when I recompiled the kernel.

Keep in mind this option doesn't persist reboots.

skgsergio, 2022-01-11:
Ok, there is a way of properly configuring it and making it persist. It requires bluez 5.55 or newer and kernel 5.9 or newer.

1. Edit `/etc/bluetooth/main.conf`
2. Add/edit the following:
   If bluez version is `5.55`:
   ```ini
   [Controller]
   LEAutoconnecttimeout=10000 # ms
   ```
   If bluez version `5.56` or greater:
   ```ini
   [LE]
   Autoconnecttimeout=10000 # ms
   ```

The limit seems to be 0x4000 (16384).

pvvx, 2022-01-31:
> I wrote to the mailing list (no response yet, but I've found later an old 2016 thread where they say what is their reason not to increase it: [link to the response](https://lore.kernel.org/linux-bluetooth/9942937A-4799-4AA2-9551-8FEBF110550B@holtmann.org/)).

The most stupid excuse (those. complete incompetence):
_the problem is that in order to send a CONNNECT_REQ, the HCI_LE_Create_Connection command needs to see connectable advertising packet one more time. So the longer time you give to HCI_LE_Create_Connection to find it, the longer everything else in the system is blocked. Since only one HCI_LE_Create_Connection can be running at the same time._

If the adapter does not allow simultaneous operation with other functions, then the connection interval is calculated by a timer.
For these adapters, the connection function must be started by a timer before receiving a new advertising package (timer - 10%).

PS: Clearly Arduno crawled into Linux.

pvvx, 2022-01-31:
TLSR825x (similar to Xiaomi LYWSD03MMC) automatically calculates the advertisement interval in 2 steps and then adjusts to switch synchronous advertisements and receive advertisements from the follower.
Consumption chart. Average 71..75 uA:
[image]

Reception window:
[image]

Discrepancies, because the ad interval should contain a random additive to reduce collisions:
[image]

henkiejan1, 2022-09-25:
> Smartphone with Android and Windows PC is always connected.

Strange enough that it also works without problems on Android because it´s a Linux based operating system....

But on my Samsung S21 with original Android 12 from Samsung disconnect from time to time. In notice also that the battery of the sensor need to be almost new when connect. Than it will work a lot better. Also on Mint 21.  

But i like your firmware very much and works really nice! Thank you. :)

closed 2023-02-18 by pvvx

pvvx, 2023-12-24:
> Strange enough that it also works without problems on Android because it´s a Linux based operating system....

“Based on” Linux – long ago rewritten by Google.

bam80, 2023-12-24:
> 2\. If bluez version is `5.56` or greater:
>         ```ini
>         [LE]
>         Autoconnecttimeout=10000 # ms
>         ```
> 
> 
> The limit seems to be 0x4000 (16384).

I didn't manage to make it work, tried values between 10000 and 16384 (I have fw advertisement interval of 10s).
There is also `ScanWindowAutoConnect` option, does it need tweaking too?