pvvx/ATC_MiThermometer #173: Flashed the wrong image. How to recover from that?

wildblue, 2022-01-03:
Iflashed my LYWSD03MMC with the wrong firmware. Now i can not do any updates on that device anymore. Is there a way to recover from that mistale?

Here is the info from the device in current state:

Status: Connecting to: Qingping Temp & RH M ...
Temp/Humi: waiting notify for data after connecting
MiVer5, DevID: 0x0B48-CGG1_ENCRYPTED, FnCnt: 1, CtrID: 0x5830 Not bound, Standard certification, MAC: 7E0CBB38C1A4, Capability: 0x28 - front binding, IO: 0x0001

pvvx, 2022-01-03:
1. Download [LYWSD03MMC Custom Firmware Version 3.5](https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_V35a.bin)
2. Manually select the downloaded **ATC_V35a.bin** file.

wildblue, 2022-01-03:
No luck. I tried all 3 telink flashers, one disconnects instantly, the other supply following error:
16:21:09: Searching for devices
16:21:14: Connecting to: Qingping Temp & RH M
16:21:14: Connected.
16:21:24: File was selected, size: 63668 bytes
16:21:24: Count: 3980
16:21:26: Start DFU
16:21:27: Disconnected.
16:21:27: Update error: some error while sending char data
16:21:32: Start DFU
16:21:33: Update error: some error while sending char data

pvvx, 2022-01-04:
The original FW CGG1 uses a button on the PC4 pin to activate it before connecting.
On LYWSD03MMC gpio pin PC4 on pcb HW B1.4 is marked "P9".

[Another option is to program via a USB-COM adapter.](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version)

wildblue, 2022-01-05:
just to confirm again:
p9 pulled to gnd before connecting?

closed 2022-05-01 by pvvx