pvvx/ATC_MiThermometer #174: Cannot connect after flash firmware and setting PIN (LYWSD03MMC)

ronald8192, 2022-01-03:
Device: LYWSD03MMC

After I flash the custom firmware, rename the device, and set the PIN, then it cannot connect anymore. 
Do I need the PIN to connect the device? I don't see the flasher webpage can enter the PIN...

Logs:
```
12:20:44 AM: Searching for devices
12:21:27 AM: Connecting to: ATC_B
12:21:36 AM: Disconnected.
12:21:36 AM: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
12:21:36 AM: Reconnect 1 from 5
12:22:17 AM: Disconnected.
12:22:17 AM: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
12:22:17 AM: Reconnect 2 from 5
12:22:22 AM: Disconnected.
12:22:22 AM: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
12:22:22 AM: Reconnect 3 from 5
12:22:39 AM: Disconnected.
12:22:45 AM: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
12:22:45 AM: Reconnect 4 from 5
12:23:00 AM: Disconnected.
12:23:00 AM: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
12:23:00 AM: Reconnect 5 from 5
12:23:08 AM: NetworkError: Connection failed for unknown reason.
12:23:08 AM: Something went wrong, too many reconnects
```

pvvx, 2022-01-03:
If you entered a pin code in TelinkMiFlasher.html, then the next connection will require you to enter a pin code.

* After setting the pin code, it is often necessary to reboot the BT adapter.

ronald8192, 2022-01-04:
I have reinstalled BT driver/reconnect the BT adapter. Still not showing up in Windows 10 new BT device scan. (only the other LYWSD03MMC with original firmware show up)

I also tried BT scan on MacBook, and Raspberry Pi using `bluetoothctl`, still cannot scan the device. The one with custom firmware but without PIN also showed up this time.

=====================
I have 3 LYWSD03MMC:
- original firmware 
- custom firmware, no PIN
- custom firmware, have PIN

pvvx, 2022-01-04:
BLE thermometer connection and scanning is not required by system tools. The connection is made only with specialized software - [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

If you need the visibility of the device in the system monitor and you do not want to save the battery, then set the option: 
[image]

ronald8192, 2022-01-04:
Found that the only thing need to do is reboot my computer. 
Reinstall BT driver, remove the device in device manager, reconnect BT hardware not work for me.

I also set PIN on another LYWSD03MMC today, reboot computer, the first time connect still fails (Windows did ask for PIN), second time connect succeeded.

```
5:52:48 PM: Searching for devices
5:52:56 PM: Connecting to: ATC_A
5:53:03 PM: NetworkError: Connection failed for unknown reason.
5:53:03 PM: Reconnect 1 from 5
5:53:52 PM: Disconnected.
5:55:22 PM: Searching for devices
5:55:29 PM: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
5:55:29 PM: Reconnect 1 from 5
5:55:52 PM: Connecting to: ATC_A
5:55:52 PM: Detected custom Firmware
5:55:55 PM: Hardware Revision String: B1.4
5:55:55 PM: Detected custom Firmware
5:55:55 PM: Detected alternative Firmware? Use: https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html
5:55:55 PM: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.5, Sensor: SHTC3
```

closed 2022-01-10 by ronald8192

Protoncek, 2022-07-17:
(after a while...)
I learned  interesting fact: 
- first i connect to xiaomi with my PC - at this point PIN is not set. Connection is succesfull.
- i set PIN code - success.
- Now i close Chrome to disconnect, reopen Chrome and try to reconnect - it won't go at all- Windows does find xiaomi and offers to enter PIN, i enter it, but connection is still UNsucsesfull.
- i grab my android phone and try the same - i can't connect either. Since i DID connect to my xiaomi without PIN before my phone has device remembered, too.
- interesting, i CAN delete my device from my phone's devices.
- now connection is succesfull when entering PIN.

So, the point is that Windows PC remembers that certain device was connected without PIN (that is, pin is 000000) and doesn't connect even if new PIN is entered. It's obviousy mandatory that this device is deleted from Win BT stack before reconnect with new PIN. 

Now, question: how do i delete known xiaomi's from Windows BT list? They are not  shown there (as is, for example my mouse, even if it's disconnected). I should have my 8 xiaomi's shown in Win BT list, but none is there...

Any clues?

ronald8192, 2022-07-20:
Not sure which Windows version you using, but for Windows 10, should be able to see the device even if disconnected.

Control panel:
* Control Panel -> All Control Panel Items -> Devices and Printers -> Under "Unspecified"

New setting app:
* Devices -> Bluetooth and other devices -> Under "Other devices"

reboot PC might also help

pvvx, 2022-09-14:
> 1. Remove other paired thermometer
>    
>    * I have other BT headphones paired but will not affect pairing PINed thermometer

BT headphones are not a BLE device, they are a Bluetooth device.