pvvx/ATC_MiThermometer #175: Bluetooth: hci0: advertising data len corrected

mvdklip, 2022-01-03:
I get a lot of "Bluetooth: hci0: advertising data len corrected" messages in /var/log/syslog and today I found out that these messages appear at exactly the same time as the advertising messages from the thermometers running pvvx/ATC firmware. Can it be that this firmware is sending incorrect data lengths which are being corrected by the bluetooth stack?

pvvx, 2022-01-04:
The size indicators in the thermometer advertisement package are correct - check the BLE sniffer.
Likewise, Wireshark does not issue any warnings when analyzing advertising packets from thermometers.
nRF Connect and other analyzing software accepts everything perfectly.
None of the logs in any system generates any warnings.

pvvx, 2022-01-04:
You have a crooked operating system - Linux.
This has been fixed long ago:
https://elixir.bootlin.com/linux/v5.10.42/source/net/bluetooth/hci_event.c#L5438

closed 2022-01-04 by pvvx

mvdklip, 2022-01-04:
The link you provided has the code in question still in place. Latest still
has it but has the logging removed:

https://elixir.bootlin.com/linux/latest/source/net/bluetooth/hci_event.c#L5611

I'm not a bluetooth expert but the comments in the code reveal a bit about
why the length would be 'wrong':

/* Find the end of the data in case the report contains padded zero
 * bytes at the end causing an invalid length value.

mvdklip, 2022-01-04:
Also:

https://devzone.nordicsemi.com/guides/short-range-guides/b/bluetooth-low-energy/posts/bluetooth-smart-and-the-nordics-softdevices-part-1

"You should set the length of the advertising data (dlen) to match with the
significant part length so that the non-significant part (zero padding)
will not be transferred over the air."

I am not saying that the firmware is sending wrong lengths. I was just
wondering if it could be. It ofcourse can also be a different device in my
neighbourhood. ;-)

pvvx, 2022-01-04:
This is the result of being received by the adapter and not by the transmitting device.