pvvx/ATC_MiThermometer #176: Qingping Temp & RH M connection issue

mmkrysiak, 2022-01-06:
Hi,
I can not connect my new Qingping Temp & RH M :(
I looks like it is not correctly recognized. Instead of CCG1 I can see CGPR1 in log

14:52:50: Connecting to: Qingping Temp & RH M
14:52:54: Hardware Revision String: 0001
14:52:54: Detected Qingping CGPR1 device
14:52:54: NotFoundError: No Characteristics matching UUID 00000017-0000-1000-8000-00805f9b34fb found in Service with UUID 22210000-554a-4546-5542-46534450464d.

pvvx, 2022-01-08:
v3.6  [TelinkMiFlasher.html ](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

mmkrysiak, 2022-01-11:
Many thanks. I can connect now. I've had a problem to activate device but with given ble key and identity I could flash without any problem.

closed 2022-01-12 by pvvx